Ultimaker/Cura #12931: Make overhang printable in spiralize outer contour mode does not work properly

fkiliver, 2022-08-02:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Voron 2.4

### Reproduction steps

1. Import a model with large angle overhang
[image]
2. Enable spiralize outer contour
3. Enable overhang printable
4. Slice

If not enable overhang printable,it like this
[image]
[image]

### Actual results

The overhang of the lower surface of the model is modified to be printable, but the upper surface is ignored
[image]

[image]

### Expected results

The overhang of the upper and lower surfaces should both be made printable
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[test.zip](https://github.com/Ultimaker/Cura/files/9243709/test.zip)

GregValiant, 2022-08-02:
Thanks for the report.
"Make Overhangs Printable" acts on overhangs.  That's why the underside was altered.  It needed support and now it doesn't.  The top surface is not an overhang and never needed support and so it isn't altered.  Spiralize never puts _top skins_ on a model and so you end up with an open space that will allow what's above to fall inside and the print will fail.
What you want to do really needs to be done in CAD.  It appears as if you want to print a different model than what you are giving Cura to work with.  Cura is not a CAD program and there is only so much modification that it can do.
What I see here is "Expected Behavior" rather than a bug.  You could submit a Feature Request (but I don't think it would get much traction with the Cura team as it starts to cross the line into CAD functions).
I'll remove the Bug label but leave this open for a bit so you can respond here.

fkiliver, 2022-08-02:
> 

Thank you for your replay.
Well, sometimes I need to print oversized sculpture models. Any filling and support will cause the cost and printing time to get out of control, so I need to use spiralize outer contour, but this will lead to the loss of support on the upper surface and the printing failure. In this case, cura just need to make the overhang printable modify the upper surface, just like what it does to the lower surface.
When the model is upside down, the upper surface becomes the lower surface. This is not the field of CAD. You can think that what I need is to combine the modification of the model by making overhang printable when the model is upright with the modification of the model by making the overhang printable when the model is upside down

GregValiant, 2022-08-03:
By definition, that upper surface is not an overhang.  Whether it is right side up, upside down, or laying on it's side, an overhang has air below and the top surface never does.  It has "part" below.
I'll close this now.

closed 2022-08-03 by GregValiant