Ultimaker/Cura #12932: Acceleration Tower slicing failed - 5.1.0

scsilvers, 2022-08-02:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender3v2

### Reproduction steps

Tried to print an acceleration tower.
Selected Extensions > Part for calibration > Add an Acceleration Tower
Tried to slice

### Actual results

Got a "Slicing Failed" error window
Slicing failed with an unexpected error.  Please consider reporting a bug

### Expected results

Tower should have slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9245685/cura.log)
[image]
[Acceleration Tower.zip](https://github.com/Ultimaker/Cura/files/9245687/Acceleration.Tower.zip)

GregValiant, 2022-08-03:
Thanks for the report.
The problem here is similar to a problem that was reported in 5.0.  It would appear to need some further work.
A workaround is to rotate the model 90° about the Z axis.

The plugin is provided by @5axes.

5axes, 2022-08-04:
The Ender-3 V2 is not in the list of the standard printer and As I don't have any issue with the Ender3 standard configuration , Your problems certainly comes from the printer definition. But I cannot analyse this issue. 
[image]

5axes, 2022-08-04:
What is the difference between an Ender-3 or an Ender-3 pro ? praticaly nothing just the size of the build plate . 220 x 220   for the Ender 3 pro ( and certainly for your Ender 3 V2 )  and 235x235 for the Ender-3 they are no other diffrence.   And I don't know why but if the size of the X plate is 220 then you have this bug ! Very very weird !  you can use a build plate with a size of 219 or 221 it works but not 220 ? 
221 : 
[image]
220 : 
[image]

5axes, 2022-08-04:
Build plate 220x220 but model size 121x121 instead of 120x120
[image]

Solved by changing the external size of the model.

GregValiant, 2022-08-04:
The size of glass on both the Ender 3 and the 3 Pro is 235 x 235.  Earlier versions of the definition files for both printers called for 220 x 220 build surface in Cura.  People complained that "they bought a 235 x 235 printer and should have it all available" so at some point it was changed to 235 x 235.  Mind you, it has a 1mm beveled edge around the glass so the "printable area" must be less that 235 x 235.
I'm with you though, I have no idea how that could affect slicing.  When I made by build plate 220 x 220 and then imported the model from the OP's project file it refused to slice.  Going back to my normal 230 x 230 there was no problem.
This appeared in my cura.log file after a failed slice:
Backend exited abnormally with return code 3221225477!

GregValiant, 2022-08-06:
I played with this some more.
If the model sits between X = -0.0005 and X = +0.0100 (inclusive) it won't slice if the bed is 220 x 220.
[GVcura.zip](https://github.com/Ultimaker/Cura/files/9274276/GVcura.zip)

5axes, 2022-08-06:
For me it's an issue regarding Cura itself. But I would be very curious to know the relation between this model and the bed size.

GregValiant, 2022-08-06:
I might be looking at this the wrong way.  #12963 appears to have the same problem but I am starting to think the issue is related to the "X Midpoint" of the bed rather than the actual size of the bed.

scsilvers, 2022-08-08:
Sorry it's taken me a few days to get back on this.  So yes, once I changed my bed settings from 220 x 220 it's slicing fine.  I also tried at the 220 x 220 bed size moving the model from the default position and that also slices ok.  So I think you're right that the bed size isn't the causal factor.

GregValiant, 2026-08-19:
This problem seems to have gone away at some point in time.  There doesn't seem to be a problem in 5.13.0 and so I'll go ahead and close this.

closed 2026-08-19 by GregValiant