Ultimaker/Cura #12933: Cura material configuration bug

Sengelon2210, 2022-08-03:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

FLSUN Q5

### Reproduction steps

1. clicked the cenfigure CURA option 
2. clicked "materials"
3. assigned and made new material profiles for a while(everythink took a long time to load"

### Actual results

after a few seconds of waiting for the material view to change to another material, CURA crashed

### Expected results

CURA should have switched the view to another material

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9248575/cura.log)

MariMakes, 2022-08-26:
Hey @Sengelon2210 ,

Welcome to Github 🎉 
Sorry it took us a while to get back to you 😞 

I'm having some trouble recreating your issue over here.
Can your share your configuration folder?
That makes it easier to get an understanding of what you are encountering,

You can share your configuration folder by:
1. Go to Help at the top bar of Cura
2. Select Show Configuration Folder
3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your 5.0 or 5.1 folders whatever you are working with.. 

Good luck!

Sengelon2210, 2022-09-04:
Hello,
Sorry it took me a while to reply
Here is the configuration folder

Thank you,
Sengelon

MariMakes, 2022-09-13:
Hey @Sengelon2210,

Did you include the file as an attachment to your email? 
Because it's not showing up on Github. 😞 

You'll need to open Github in your browser to attach any files.
You can click the **view it on GitHub** link at the bottom of your email to open Github in the browser. 

Can you please try to share your configuration again?
Thank you 😇

Sengelon2210, 2022-09-14:
hey @MariMakes  
i am including the file throught github now, last time i attached the file in outlook

i also noticed that i cannot disable the prime blob, i can turn it off in prefferences, but it is never off/ shows up in the print settings on the right.

[configuration folder.zip](https://github.com/Ultimaker/Cura/files/9567436/configuration.folder.zip)

closed 2022-11-16 by github-actions[bot]