Ultimaker/Cura #12934: Flow Rate Problem

cemalcraycle, 2022-08-03:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Normally sliced a model with my standard settings.

### Actual results

Flow rate is almost zero.

### Expected results

I have checked startcode, material settings and there is no problem.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I was using Cura 4.*.* and there was no problem with same model and same settings profile. Someting is wrong and sets the flow rate low.

GregValiant, 2022-08-03:
We need a project file to investigate.  Load the model and use the "File | Save Project" command.  Zip the 3mf project file and post it here on the site.

Asterchades, 2022-08-04:
Double check that your printer is still configured to use the correct filament diameter. It wouldn't be the first time I've seen it that this setting gets reverted to 2.85mm instead of 1.75mm, which would effectively reduce the flow rate by approximately 62% relative to however it is configured.

From the Preferences menu select Configure Cura. Navigate to the Printers section, highlight your printer (if there is more than one listed) and select Machine Settings. On the Extruder tab ensure that Compatible Material Diameter is properly configured.

cemalcraycle, 2022-08-05:
I have checked and yes filament diameter is wrong. It's 2,85 but I use 1,75 mm filaments. Everything is fine right know and thank you for your helps! Sorry for missing simple detail :(

closed 2022-08-22 by cemalcraycle