Ultimaker/Cura #12936: Manage Profile button missing

dynamarc, 2022-08-03:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker 5S

### Reproduction steps

In Print setting i clicked on a specific profile 

### Actual results

At the bottom of the dropdown list  the Manage Profile button is abscent

.

### Expected results

I expect to seed the Manage Profile button like in cura 4.12.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additionnal info

jellespijker, 2022-08-03:
We can only help you if you provide us with the log and/or project files.

The log file, located at:
- Windows: `%APPDATA%\cura\<Cura version>\cura.log` or usually `C:\Users\<your_username>\AppData\Roaming\cura\<Cura version>\cura.log`
- MacOS: `$HOME/Library/Application Support/cura/<Cura version>/cura.log` (OSX)
- Linux: `$HOME/.local/share/cura/<Cura version>/cura.log`

MariMakes, 2022-08-26:
Hey @dynamarc,

Welcome to Github! 🎉 
If you want to share files on Github you'll have to upload them in the browser.
The file in your email did not arrive.

I'm having some trouble reproducing your issue here. 
[image]
Can you share a screenshot or even a video of what you are encountering?

At first glance, it looks similar to https://github.com/Ultimaker/Cura/issues/12790
Can you let me know if it's a duplicate and if the provided workaround works for you?

Thank you!

closed 2022-09-09 by github-actions[bot]