Ultimaker/Cura #12937: Cura is capturing my Window manager Shorctus to tile the window

leoheck, 2022-08-03:
### Application Version

5.1.0

### Platform

Ubuntu 22.04

### Printer

-

### Reproduction steps

Flag + Arrows does not do anything on cura. I cannot tile the window. 

### Actual results

nothing happpens

### Expected results

cura is tiled to the sides of the monitor.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

jellespijker, 2022-08-03:
We can only help you if you provide us with the log and/or project files.

The log file, located at:
- Windows: `%APPDATA%\cura\<Cura version>\cura.log` or usually `C:\Users\<your_username>\AppData\Roaming\cura\<Cura version>\cura.log`
- MacOS: `$HOME/Library/Application Support/cura/<Cura version>/cura.log` (OSX)
- Linux: `$HOME/.local/share/cura/<Cura version>/cura.log`

leoheck, 2022-08-03:
Something was weird, it started work after restarting the computer. Sorry for the inconvenience.

closed 2022-08-03 by leoheck