Ultimaker/Cura #12939: Slicing error

bojangles86, 2022-08-03:
### Application Version

5.1.0

### Platform

Linux Garuda

### Printer

Ender 5

### Reproduction steps

Tried slicing .stl 
recieved error "slicing failed"
Turned off extensive stitching and tried again.
recieved same error.
most other settings are defaulted besides temps. 
[image]

### Actual results

tried changing settings to default and tried slicing again

### Expected results

expected .stl to slice but get error instead

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9252806/cura.log)

jellespijker, 2022-08-03:
We really need the 3mf project file as well,l. It will be nearly impossible to debug and reproduce this issue otherwise

bojangles86, 2022-08-03:
[CE5_Laser_Cat_-_Voronoi_fine.zip](https://github.com/Ultimaker/Cura/files/9255260/CE5_Laser_Cat_-_Voronoi_fine.zip)

sorry i thought i had uploaded this.

GregValiant, 2022-08-03:
Win10 Cura 5.1.  I opened it and it sliced.
(I would have lost a bet with myself on that as I was sure there would be model errors.)
Will it print?  Now that is a cat of a different color.
[image]

MariMakes, 2022-08-26:
I can also slice your project file in Cura 5.1
[image]

If it's not an issue anymore, can we close the issue?

bojangles86, 2022-08-26:
Sure. I think it's safe to close now.

closed 2022-08-26 by MariMakes