Ultimaker/Cura #12940: Cura 5 appimage (multiple versions) does not open on linux

gazhay, 2022-08-03:
### Application Version

5.0

### Platform

Ubuntu 20.04

### Printer

Overlord Dreamamker

### Reproduction steps

Download,
Mark as executable
Run

You see splash screen and it goes away, no other windows ever open

14.x versions work fine.

I do have multiple displays under NVIDIA drivers, which is potentially an issue with windows users, but I couldn't see one for linux.

### Actual results

Nothing - cura does not open

### Expected results

Cura should open like the 14.x versions.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No idea where the appimage writes its log file, not in the usual places.

jellespijker, 2022-08-03:
It looks like you tried Cura 5.0. Although the multiple versions remark is a bit confusing.
Can you try the 5.1 version designated as `modern` if that doesn't work try the regular 5.1 AppImage.

gazhay, 2022-08-03:
Its anything since the jump to version 5. Including Betas (which is a reason I didn't report earlier) 

The bug is present in the currently available download. 

The modern variant doesn't even get to the splash screen, console error 

````
[29651] Error loading Python lib '/tmp/.mount_Ultima8sQDKJ/libpython3.10.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /tmp/.mount_Ultima8sQDKJ/libpython3.10.so.1.0)
````

gazhay, 2022-08-03:
The non modern variant returns 

````
QApplication: invalid style override passed, ignoring it.
    Available styles: Windows, Fusion
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
qt.gui.icc: fromIccProfile: failed size sanity 2
qt.gui.icc: fromIccProfile: failed size sanity 2
qt.gui.icc: fromIccProfile: failed size sanity 2
QApplication: invalid style override 'gtk' passed, ignoring it.
	Available styles: Windows, Fusion
load glyph failed err=6 face=0x23f4600, glyph=65535
QQmlApplicationEngine failed to load component
QQmlComponent: Component is not ready
````

jellespijker, 2022-08-03:
Can you add the log files as well

The log file, located at:
- Windows: `%APPDATA%\cura\<Cura version>\cura.log` or usually `C:\Users\<your_username>\AppData\Roaming\cura\<Cura version>\cura.log`
- MacOS: `$HOME/Library/Application Support/cura/<Cura version>/cura.log` (OSX)
- Linux: `$HOME/.local/share/cura/<Cura version>/cura.log`

gazhay, 2022-08-04:
[cura.log](https://github.com/Ultimaker/Cura/files/9259798/cura.log)

This is from the 5.1 directory

lokcon, 2022-08-09:
Running into the same issue (although running on Arch Linux and Wayland/sway): splash screen shows and goes away, then nothing happens.

Running:
`LD_PRELOAD=/usr/lib/libstdc++.so.6 QT_QPA_PLATFORM=xcb ./Ultimaker-Cura/Ultimaker-Cura-5.1.0.AppImage`

yields:
```
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
QApplication: invalid style override 'kvantum' passed, ignoring it.
	Available styles: Windows, Fusion
QQmlApplicationEngine failed to load component
QQmlComponent: Component is not ready
```

lokcon, 2022-08-09:
Managed to start Cura after a little bit of digging. The issue seems to be related to custom Qt theme engine, and `QT_STYLE_OVERRIDE` somehow.

Looking at `~/.local/share/cura/5.1/cura.log` after a failed start, the last error in the log for me was:

`2022-08-09 20:33:37,747 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [617]: file:///tmp/.mount_UltimaqNlLE3/share/cura/plugins/SliceInfoPlugin/MoreInfoWindow.qml: module "kvantum" is not installed`

I do have [kvantum](https://github.com/tsujan/Kvantum) installed on my system; The [Arch Linux wiki](https://wiki.archlinux.org/title/Uniform_look_for_Qt_and_GTK_applications#Kvantum) suggests using `QT_STYLE_OVERRIDE=kvantum` to enable kvantum, which was the case in my bashrc.

*I then set `QT_STYLE_OVERRIDE=""` in front of `./Ultimaker-Cura/Ultimaker-Cura-5.1.0.AppImage` in the command line and Cura started successfully^.*

^For me on Arch and Wayland, the complete command in the end was: `LD_PRELOAD=/usr/lib/libstdc++.so.6 QT_QPA_PLATFORM=xcb QT_STYLE_OVERRIDE="" cura
`

gazhay, 2022-08-22:
Can confirm https://github.com/Ultimaker/Cura/issues/12940#issuecomment-1209244054

works on ubuntu 22.04.01 LTS

MariMakes, 2022-08-26:
Hey @gazhay,

I might have read it incorrectly but it seems like your issue has been resolved. 
If so can we close this issue?

gazhay, 2022-08-26:
Well, its "solved" in that I can launch it with q clunky command line. Fine for me, but if the bug affects more than just me it probably needs addressing in the source. I e. Set QT_STYLE_OVERRIDE somewhere.

lpinilla, 2022-09-12:
Yeah, same problem on Ubuntu 22.04, although and I can't use the fix because I don't have `/usr/lib/libstdc++.so.6` even after installing `libstdc++6` via `apt install`.

Kieffer87, 2022-10-10:
Any update on this for ubuntu 22.04?

abo1787, 2023-01-01:
I'm having a similar issue with all 5.0 and greater packages with all versions of the appimage.  I cannot get any of them to launch due to glibc errors.  I'm on Ubuntu 18.04 LTS, and sadly cannot upgrade as it's a work machine.

jellespijker, 2023-01-01:
@abo1787 Unfortunately we no longer support Ubuntu18.04 This is because of our PyQt6 upgrade which simply required a higher glibc version. Unfortunately, we can do anything about it.

abo1787, 2023-01-03:
Thanks for the info man.  That sucks, but I do see that dell has certified my laptop for 20.04.  I guess now that I know what the package causing me issues is I can find out if it's supported on 20.04 and if that'll fix my issue.  Looks like glibc 2.31 is what is bare minimum required for pyqt6.  I'll roll the dice with a live image, and hopefully that fixes my problem :)