Ultimaker/Cura #12941: Build instructions are outdated and don't work

richfelker, 2022-08-03:


richfelker, 2022-08-03:
I wasn't aware of the additional instructions in the Cura wiki. I'll see if I can extract the information I need from there. I'm not trying to install the Cura frontend only build CuraEngine, and it's not clear right away how to do that. The instructions in the CuraEngine README no longer work, as clipper and other dependency code has been removed and no instructions on how to get it have been added.

richfelker, 2022-08-03:
OK, trying to follow those directions has Conan downloading hundreds of MBs of packages that duplicate basic system functionality/libraries I already have and want to use rather than just getting the missing clipper, stb, etc...

jellespijker, 2022-08-03:
You will only have to build those packages such as: boost, clipper, Arcus etc and those will be done automatically. Once they are build they will then be stored in the local Conan cache and only updated or rebuild if they're changed, or no longer ABI compatible.

Yes it is a bit of a change but compared to the old workflow we have shaved down build time by a lot. It is possible to setup a working development environment within 2 minutes instead of a day (no exaggeration btw)

We now also have reproducable builds, no longer a dependency hell, which could cause bugs and eat away time and resources from developers trying to figure out what the difference is between systems. While finding out it was a different version of an indirect dependencies.

Not to mention that we have a cost reduction in the ball park of 1 senior developer on a yearly basis.

I know it is a change, but we honestly think this is already a huge improvement on what we had and I hope you guys can give use feedback what works for you and what can be improved. Because one of reasons we switched is that it should greatly reduce the complexity for developers to set up the environment.

Please try out the work flow for a bit, see if it is an improvement and give us feedback or PR's if you think that somethings need to be changed.

jellespijker, 2022-08-03:
Once Conan is installed the only instructions for building CuraEngine are

```
conan install . -pr:b cura_build.jinja -pr:h cura_release.jinja --build=missing
cmake . -DCMAKE_TOOLCHAIN_FILE=cmake-build-release/conan/conan_toolchain.cmake
cmake --build .
```

jellespijker, 2022-08-03:
@rburema can you update the README.md for Curaengine as part of ticket CURA-8830

It is indeed outdated

Piezoid, 2022-08-03:
> `conan install . curaengine/5.2.0-alpha@richfelker/testing --build=missing --update`

This fails for me with `ERROR: Package recipe with name curaengine!=cura-build-environment`

<details>

```shell

$ conan --version
Conan version 1.51.0

$ conan profile show default
Configuration for profile default:

[settings]
os=Linux
os_build=Linux
arch=x86_64
arch_build=x86_64
compiler=clang
compiler.version=14
compiler.libcxx=libstdc++
build_type=Release
[options]
[conf]
[build_requires]
[env]

$ find /home/piezo/.conan/  
/home/piezo/.conan/
/home/piezo/.conan/.gitignore
/home/piezo/.conan/remotes.json
/home/piezo/.conan/global.conf
/home/piezo/.conan/.conan.db
/home/piezo/.conan/artifacts.properties
/home/piezo/.conan/profiles
/home/piezo/.conan/profiles/default
/home/piezo/.conan/profiles/cura_build.jinja
/home/piezo/.conan/profiles/cura.jinja
/home/piezo/.conan/version.txt
/home/piezo/.conan/generators
/home/piezo/.conan/generators/VirtualPythonEnv.py
/home/piezo/.conan/generators/PyCharmRunEnv.py
/home/piezo/.conan/generators/GitHubActionsBuildEnv.py
/home/piezo/.conan/generators/GitHubActionsRunEnv.py
/home/piezo/.conan/cacert.pem
/home/piezo/.conan/settings.yml
/home/piezo/.conan/conan.conf
/home/piezo/.conan/config_install.json
```
</details>

jellespijker, 2022-08-03:
@Piezoid it looks like that you're running that from the cura-build-environment repo, which is deprecated.

The Conan install command for Curaengine should be run in the root of Curaengine.

The biggest change in our workflow is that we no longer rely on long complex interdependent build scripts, but that each repo can now be seen as a single modular project. I hope this documentation explain the philosophy best https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#process-description-and-workflow

jellespijker, 2022-08-03:
@Piezoid as described earlier in this thread, our old build process wasn't maintainable anymore and very very costly both money wise and on development hours.

> It used to be possible to build cura and its dependencies in one command

It is now possible the install the latest Cura (main) in one line (once Conan etc is setup)
```
conan install cura/latest@ultimaker/testings --build=missing --update -g VirtualPythonEnv -o de tools=True 
```

The aim is to make this workflow easy for all OSes, build systems and development machines.
I can currently setup a whole development environment within minutes, and we can build an AppImage within 6 minutes.

Yes at the moment you still can encounter some issues, because we are still working on setting things up. But we are confident that we will get there within the next two weeks.

jellespijker, 2022-08-03:
@Piezoid 
> And what if I want to use a locally installed Arcus, since I also have changes in local source?

If you want to develop multiple projects like for instance arcus and Curaengine. Clone them both and then put Arcus into editable mode https://docs.conan.io/en/latest/developing_packages/editable_packages.html

```
cd libArcus
conan editable . arcus/5.2.0-alpha@piezod/Ultimaker
cd ../Curraengine
Conan install . curaengine/5.2.0-alpha@ultimaker/testing --require-override=arcus/5.2.0-alpha@piezod/Ultimaker --build=missing --update
cmake --preset Release # for Cmake version 3.23+ otherwise the regular Cmake configure/ generate commands give earlier 
```

What that basically does is to override the arcus package specified in `conandata.yml` it will then use that previously created editable package which will be used to generate the CMake find modules using the paths pointing to your local Arcus repo. Just make sure that you build the actually binary of Arcus for the release build

jellespijker, 2022-08-03:
@Piezoid it really looks like you're are using the `cura-build-environment` repo. That one is no longer used.

Piezoid, 2022-08-03:
> The aim is to make this workflow easy for all OSes, build systems and development machines. I can currently setup a whole development environment within minutes, and we can build an AppImage within 6 minutes.

I get that it greatly simplifies the process on your end for CI and releasing artifacts. However, maintainers of linux packages will need to somehow integrate this in their workflow and build against their own packages. Shipping a separate build of openssl, Qt, etc is not acceptable on these channels. Maybe there is way around this, but the fact that conan-io/conan#1330 is still open is not encouraging.

I will take a closer look at the instructions, the editable mode seems to cover my needs for development.

richfelker, 2022-08-03:
> Hundreds of mb sounds like it is indeed installing the frontend with cpython etc. For Curaengine you would only have to install Conan and the Conan configuration.
> 
> ```
> conan config install https://github.com/ultimaker/conan-config.git
> conan profile new default --detect --force
> ```

Can you clarify what these commands are supposed to be doing and where they're supposed to be run from? Are they setting up some user-account-wide conan configuration imported from your github, or are they project-local?

> Once Conan is installed the only instructions for building CuraEngine are
> 
> ```
> conan install . curaengine/5.2.0-alpha@richfelker/testing --build=missing --update 
> cmake . -DCMAKE_TOOLCHAIN_FILE=cmake-build-release/conan/conan_toolchain.cmake
> cmake --build .
> ```

What is "`curaengine/5.2.0-alpha@richfelker/testing`"? I am not trying to pull CuraEngine source from my github. I'm trying to use my local repository with the commits I want in it. When I tried to do something like this before, it started downloading all sorts of unrelated stuff under `~/.conan`.

jellespijker, 2022-08-03:
> How can I compile using system zlib, protobuf and boost packages?

Why would you want to do that?

If you would really want to do that you would have to remove those requirements out of the `conandata.yml` and make sure that CMake module path has find modules with the same namings scheme and which declare the same cmake target names.

Which is a pain and this is mainly one of the reasons why we moved to Conan. Maintaining Cura with packaged installed via multiple different system package manager (or OSes without a package manager Windows) is a pain, error prone and filled up the swear-jar very quickly.

The benefit of using Conan is that binaries are reused for every project managed by Conan. Depending on the user options, compiler, architecture and os. See also https://docs.conan.io/en/latest/getting_started.html#installing-dependencies

richfelker, 2022-08-03:
> Why would you want to do that?

Because I don't want N copies of these things installed, and this is the way we do things when we're compiling a package to ship in a distro or to run on our own systems: we use the distro-provided libraries we already have rather than every program having its own pet way to get yet another copy of the same thing.

> If you would really want to do that you would have to remove those requirements out of the `conandata.yml` and make sure that CMake module path has find modules with the same namings scheme and which declare the same cmake target names.

Being unfamiliar with Conan, I'm not sure what I need to do here. There are a bunch of different headings "None", "5.1.0", etc. all with the same deps under them. I am not trying to make an artifact for distribution so I do not care about naming versions; I will be building from "git mainline" all the time, and just want to be able to build current source without downloading any library code except what's specific to CuraEngine. (I don't even need Arcus, I disable it at build time).

jellespijker, 2022-08-03:
> Can you clarify what these commands are supposed to be doing and where they're supposed to be run from?

These need to be run once and can run from everywhere. The will pull in the Conan configuration to make sure that all projects are build with the same flags, environment settings etc. You will also add a link to the cura-ce Artifactory server. Which is a server containing the conanfile.py for dependencies, also known as recipes. These contain the information to build the dependencies and how to consume it.

All that information is used to generate, in our case CMake find configuration files` but you can also output files for xcode, msbuild, auto tools etc.

I forgot that you guys need to remove the cura Artifactory remote which requires credentials and is used by us internally. This contains the recipes and pre built binaries to speed up the build steps. Unfortunately it is to costly to also allow an arbitrary number of community developers to download those.

```
conan remote remove cura
```

> What is "curaengine/5.2.0-alpha@richfelker/testing"? I am not trying to pull CuraEngine source from my github.
Notice the `.` in the `conan install` command which specifieke that we're using the conanfile.py in the root of CuraEngine. So you're building from the local repo.

> When I tried to do something like this before, it started downloading all sorts of unrelated stuff under ~/.conan

It should only download, dependencies and dependencies from dependencies. That is the whole idea about Conan

You can use the following commands to get some extra insights

```
conan

jellespijker, 2022-08-03:
Sorry accidentally closed it, typing it all on a phone. Also apologizing for the "autocorrect" words and capitalization you might encounter in my comments

richfelker, 2022-08-03:
OK, from what I can tell of [the docs](https://docs.conan.io/en/latest/reference/commands/consumer/config.html):

> The config install is intended to share the Conan client configuration. For example, in a company or organization, is important to have common settings.yml, profiles, etc.

So on the one hand folks are telling me that Conan is supposed to be a system for building "every project managed by Conan", but on the other hand you're giving me instructions that, as far as I can tell, are basically "clobber any Conan configuration I had and clone Ultimaker's organizational configuration onto my system". This really does not make sense, and reinforces my frustration that Conan seems to be designed around organizational/enterprise use, not as a system for consumers of FOSS to build it on their own systems with their own user accounts.

jellespijker, 2022-08-03:
> Because I don't want N copies of these things installed, and this is the way we do things when we're compiling a package to ship in a distro or to run on our own systems:

This is what Conan tries to solve, while taking into account ABI and API compatibility.

If you want to deploy stuff to the userspace you can add `-g deploy` https://docs.conan.io/en/latest/reference/generators/deploy.html?highlight=deploy

If you want to run it directly you can use the `-g VirtualRunEnv` this will create a shell script which will set all the environment variables like `LD_LIBRARY_PATH`, `PATH` etc. Such that your project can find those installed in the local Conan cache.

jellespijker, 2022-08-03:
> I will be building from "git mainline" all the time, and just want to be able to build current source without downloading any library code except what's specific to CuraEngine

```
git clone https://github.com/ultimaker/curaengine
cd curaengine
conan install . curraengine/5.2.0-alpha@ultimaker/testings --build=missing --update
cmake --preset Release
cmake --build --preset Release
```
Should be enough

Piezoid, 2022-08-03:
> Why would you want to do that?

Mainly, I'm anticipating the requirements that will come from linux distribution packagers. They will either come ask these question or give up.

But also because I'm still struggling to have conan build all the dependencies. Running this in Cura source directory:
```
CONAN_SYSREQUIRES_MODE=enabled conan install . cura/5.2.0-alpha@piezo/stable --build=missing --update -o cura:devtools=True -g VirtualPythonEn
```
fails while building tcl/8.6.10, while I have a perfectly good version on my system.

I had to add to specify `CONAN_SYSREQUIRES_MODE` because the `verify` default mode can't find the X11 libraries that are nevertheless present (with system includes).
Then, Tcl build fails to link math function as if tclsh was not linked with libm. 
I'll try again tomorrow with gcc/binutils instead of llvm/llvm-ld.

jellespijker, 2022-08-03:
> on the other hand you're giving me instructions that, as far as I can tell, are basically "clobber any Conan configuration I had and clone Ultimaker's organizational configuration onto my system".

You can install the Ultimaker configuration to an other directory and set an environment variable to use that. They took that into account. We simply didn't get to writing the complete documentation yet for Cura and Curraengine. Since we're still in the process of switching to this workflow

richfelker, 2022-08-03:
After removing the gratutious system lib deps from `conandata.yml`, it was still trying to download a huge pile of duplicate system libs as indirect deps of Arcus. After adding `-o enable_arcus=False` to the `conan install` command, it actually succeeded, but now I don't know how to get it to build. CMake is complaining that

```
  Target "_CuraEngine" links to:

    boost::boost
```

so I think there is inconsistent naming going on in how the package is referred to in CMake, or something was broken that prevents the system boost from being found like it was before the Conan changes..

richfelker, 2022-08-03:
> You can install the Ultimaker configuration to an other directory and set an environment variable to use that.

OK, great, but the docs should be **warning** people that they need to do this, and telling them how! Since, as written, they instruct the user to clobber their own Conan configuration if they already have/use Conan.

jellespijker, 2022-08-03:
@Piezoid can you share your CMD line output maybe increase the log level to debug first:
```
export CONAN_LOGGING_LEVEL=debug
```

richfelker, 2022-08-03:
And, capitalizing Boost properly makes it work again...

jellespijker, 2022-08-03:
@richfelker agreed and we will add those instructions but I hope you also understand why that documentation is currently not there yet.
We're still busy working on this, while also trying to maintain our regular release schedule.

It will be added, just know that we are pressed for resources atm so any help in this regard is very much appreciated and please know that we didn't switch to a new system to frustrate you guys (whom we really appreciate).
The old system just wasn't maintainable anymore.

richfelker, 2022-08-03:
Full recipe that works for me (if I didn't miss anything):

```
cd /path/to/CuraEngine
conan remote add cura-ce  https://ultimaker.jfrog.io/artifactory/api/conan/cura-community True
CONAN_REVISIONS_ENABLED=1 conan install -o enable_arcus=False . --build=missing --update
sed -ie 's/boost::boost/Boost::boost/g' CMakeLists.txt
mkdir build
cd build
cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake-build-releas
e/conan/conan_toolchain.cmake -DCMAKE_BUILD_TYPE=Release -DENABLE_ARCUS=OFF  -DCMAKE_CXX_FLAGS="-static -O2"  
make
```

The `CMAKE_CXX_FLAGS` are my own preference and non-essential.

jellespijker, 2022-08-03:
> boost::boost
so I think there is inconsistent naming going on in how the package is referred to in CMake

The CMake target namings follow the naming scheme of Conan (which is lowercase by default)

We had to make a decision, and our old CMake scripts were even more inconsistent in their naming scheme. Keep in mind that Linux is just one of the three operating system on which we need to build stuff

richfelker, 2022-08-03:
Can you add the alias (which you seem to have had before) so that it works either way? I tried re-adding that but I don't know cmake well enough to make it work so I just changed it.

jellespijker, 2022-08-03:
For the CXX flags you can edit your profile btw
https://docs.conan.io/en/latest/reference/profiles.html#variable-declaration

richfelker, 2022-08-03:
And one more broken thing it wasn't doing before: it's putting `/usr/lib/libgomp.so` on the link command line rather than `-lgomp` so it tries to link a shared library into a static link. I have no idea where that wrong hard-coded path is coming from (maybe Conan's doing?) so I guess that's one more thing to track down...

jellespijker, 2022-08-03:
> Can you add the alias
I will discuss this with my colleagues @rburema tomorrow or Friday, since he is currently working on finishing up the curaengine repository.

But to be honest I myself am a bit reluctant, since that would open up other exceptions in the CMake scripts as well and experience thought us that complex CMake scripts, will get more complex overtime and in the end nobody knows anymore what is happening in that script and why it is that turning one knob there results in a blinking light over there.

I'm thinking of a way to facilitate this while keeping away from messing around in the CMale scripts.

Knowing the exact reason why you would want this would help a lot. Is the reason purely that you want to use your system boost version?
If so won't it be enough to use the exact version which is installed on your system?
It's header only after all.

You can override the version for a dependency with `--require-override=boost/1.70.0@_/_` for instance you can look up al the versions in the Conan center index
 https://conan.io/center/

richfelker, 2022-08-03:
> Knowing the exact reason why you would want this would help a lot. Is the reason purely that you want to use your system boost version?

Because installing a second Boost via Conan is an **utterly gigantic** dependency tree of stuff for Conan to fetch and build, and consequently for stuff to go wrong, and I am not going to manage that as part of trying to build and contribute to a single program.

jellespijker, 2022-08-03:
> And one more broken thing it wasn't doing before: it's putting /usr/lib/libgomp.so on the link command line rather than -lgomp

As far as I know imp is still using the CMake find_package defined by the system and is currently not managed by Conan. Maybe @rburema can investigate this?

I believe we set the https://cmake.org/cmake/help/latest/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html

Piezoid, 2022-08-03:
> @Piezoid can you share your CMD line output maybe increase the log level to debug first:
> 
> ```
> export CONAN_LOGGING_LEVEL=debug
> ```

@jellespijker [build log](https://github.com/Ultimaker/Cura/files/9255181/conan-install-log.txt)
This is using some custom compiler environment variables (thin-lto, llvm cache, etc), so it's probably not worth your time to investigate this.

Good news, I managed to `conan install` after clearing environment (now using gcc) [conan-install-log-gcc.txt](https://github.com/Ultimaker/Cura/files/9255278/conan-install-log-gcc.txt).

jellespijker, 2022-08-03:
@richfelker I know you would then have multiple boost instance on your system. But they're are installed in such away that they don't mess up your system. Conan dependencies, no system wide installation are done for these dependencies. This setup simply eans that we can guarantee ABI and API compatibility while having the option to reuse binaries.

Piezoid, 2022-08-03:
btw, now that CuraEngine is no longer using OpenMP, the flags in `conanfile.py` are probably not required, as is the dependency mentioned in the README.

richfelker, 2022-08-03:
@jellespijker: I'm not worried about them messing up my system. I'm worried about consuming vast amounts of disk space and hitting random build problems from Conan fetching hundreds of dependencies and trying to build them. I do not want to do that. My distro already did that work for me, which I greatly appreciate. I static link so that there are no ABI tracking issues.

richfelker, 2022-08-03:
OK, it's also building clipper as a shared library and failing to link that...

jellespijker, 2022-08-03:
@Piezoid good point about the openmp option.
Should be removed indeed.

jellespijker, 2022-08-03:
If you want to build it all statically just append the Conan install command with: `-o *:shared=False` (I think... I`m doing this from memory on a phone while getting ready to go to bed 😉)

jellespijker, 2022-08-03:
> My distro already did that work for me, which I greatly appreciate
I also really likey distro. But I also known that the version on my distro for boost isn't the same version that Cura uses.

jellespijker, 2022-08-03:
Just to clarify... Because this is text after all. I really appreciate you guys input and we will put effort into it, such that you guys can develop without to much pain

richfelker, 2022-08-04:
OK, as far as I can tell Conan isn't even downloading clipper source and compiling it; it seems to be using Ultimaker's binary artifacts which are only going to be valid for whatever target something in this mess is assuming I'm using (which I'm not). Is there some way to get Conan to actually download the source and build it, or is this just designed to be a binaryware ecosystem?

richfelker, 2022-08-04:
OK, that was the `--build=missing`. Using `--build` instead makes it build but still has the wrong identifying info. I think I need custom settings to avoid that, per [this answer on SO](https://stackoverflow.com/a/66492385/379897).

And I still can't find any way to get it to build static linkable code. Clipper's `CMakeLists.txt` has logic for switchable `BUILD_SHARED_LIBS` in it, but Conan isn't passing that in...

richfelker, 2022-08-04:
OK, I ended up just building clipper myself from upstream - as it turns out, upstream's build system is bogus and can't build a static library, and apparently CuraEngine was previously doing its own hack to just link the single `.o` file. So I put that in a `.a`. And just conditioned out the whole `find_package` section of `CMakeLists.txt` and added a single `target_link_libraries(_CuraEngine PRIVATE polyclipping)` which just passes `-lpolyclipping` on the link command line and lets the compiler toolchain do its normal search, in the paths I expect it to be searching, without Conan or CMake interfering. So, this is a sufficiently maintainable, zero pain solution for me now that I know how to do it. I'm still happy to try to work with upstream folks to get this more friendly to users trying to follow your official build procedures. Maybe there's some way we could document what I'm doing for folks who want to do that (and for distro packagers, who need to do that) as well?

Piezoid, 2022-08-04:
When building CuraEngine from source, I get this error: `This file was generated by an older version of protoc which is incompatible with your Protocol Buffer headers. Please regenerate this file with a newer version of protoc.`

The *.pb.[ch] files are generated with conan's protoc (3.17.1), but the C++ compilation is using system headers (3.21.4)
[`make VERBOSE=1` log](https://github.com/Ultimaker/Cura/files/9257285/protobuf.txt)

richfelker, 2022-08-04:
For the record, here's the patch that got it to work for me, building with dependencies already present in include/library path. Apologies for the naming which of course isn't appropriate for upstream - I was a bit fed up with stuff when I finally got it working last night. ;-)

```

diff --git a/CMakeLists.txt b/CMakeLists.txt
index a29716d31..0a0dfa8ea 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -187,6 +187,7 @@ if(${EXTENSIVE_WARNINGS})
     set_project_warnings(_CuraEngine)
 endif()
 
+if(ENABLE_BARBARIC_BS)
 if(ENABLE_OPENMP)
     find_package(OpenMP REQUIRED)
     target_link_libraries(_CuraEngine PUBLIC OpenMP::OpenMP_CXX)
@@ -202,6 +203,9 @@ find_package(stb 20200203 REQUIRED)
 find_package(Boost 1.78.0 REQUIRED)
 
 target_link_libraries(_CuraEngine PRIVATE clipper::clipper rapidjson::rapidjson stb::stb boost::boost)
+else()
+target_link_libraries(_CuraEngine PRIVATE polyclipping)
+endif()
 
 if (WIN32)
     message(STATUS "Using windres")
```

Piezoid, 2022-08-05:
Weirdly enough, I cannot reproduce the build error related to protobuf that I was having yesterday. I ran `git clean -fxd`, so that might be why...

> If you want to develop multiple projects like for instance arcus and Curaengine. Clone them both and then put Arcus into editable mode https://docs.conan.io/en/latest/developing_packages/editable_packages.html

@jellespijker I get a link error when using an editable Arcus package.

This works:
```bash
cd libArcus
conan create . arcus/5.2.0-alpha@piezo/testing
cd ../CuraEngine
conan install . curaengine/5.2.0-alpha@piezo/testing --build=missing --update --require-override=arcus/5.2.0-alpha@piezo/testing
conan build .
```

But this fails to link libArcus.so:
```bash
cd libArcus
conan editable add . arcus/5.2.0-alpha@piezo/testing
cd ../CuraEngine
conan install . curaengine/5.2.0-alpha@piezo/testing --build=missing --update --require-override=arcus/5.2.0-alpha@piezo/testing
conan build .
```

I noticed that the file `CuraEngine/cmake-build-release/conan/arcus-release-x86_64-data.cmake` is different in editable mode:
```diff
@@ -8,11 +8,11 @@ set(protobuf_FIND_MODE "NO_MODULE")
 
 ########### VARIABLES #######################################################################
 #############################################################################################
-set(arcus_PACKAGE_FOLDER_RELEASE "")
+set(arcus_PACKAGE_FOLDER_RELEASE "/home/piezo/.conan/data/arcus/5.2.0-alpha/piezo/testing/package/1de99f51f516481dcd11f9c7d1c1bf1460e6b519")
 set(arcus_BUILD_MODULES_PATHS_RELEASE )
 
 
-set(arcus_INCLUDE_DIRS_RELEASE "/home/piezo/code/libArcus/./include")
+set(arcus_INCLUDE_DIRS_RELEASE "${arcus_PACKAGE_FOLDER_RELEASE}/include")
 set(arcus_RES_DIRS_RELEASE )
 set(arcus_DEFINITIONS_RELEASE )
 set(arcus_SHARED_LINK_FLAGS_RELEASE )
@@ -21,9 +21,9 @@ set(arcus_OBJECTS_RELEASE )
 set(arcus_COMPILE_DEFINITIONS_RELEASE )
 set(arcus_COMPILE_OPTIONS_C_RELEASE )
 set(arcus_COMPILE_OPTIONS_CXX_RELEASE )
-set(arcus_LIB_DIRS_RELEASE "/home/piezo/code/libArcus/build/Release/.")
-set(arcus_LIBS_RELEASE )
-set(arcus_SYSTEM_LIBS_RELEASE )
+set(arcus_LIB_DIRS_RELEASE "${arcus_PACKAGE_FOLDER_RELEASE}/lib")
+set(arcus_LIBS_RELEASE Arcus)
+set(arcus_SYSTEM_LIBS_RELEASE pthread)
 set(arcus_FRAMEWORK_DIRS_RELEASE )
 set(arcus_FRAMEWORKS_RELEASE )
-set(arcus_BUILD_DIRS_RELEASE )
+set(arcus_BUILD_DIRS_RELEASE "${arcus_PACKAGE_FOLDER_RELEASE}/")
```

Putting back `set(arcus_LIBS_RELEASE Arcus)` allows to link again in editable mode, but this is not ideal.

jellespijker, 2022-11-02:
We have updated the build instructions a while ago. See https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source/  and https://github.com/Ultimaker/Cura/wiki/Build-With-GitHub-Actions

I'm closing this issue as completed.

Since there was also a lot of semi-related discussion about using Conan as a dependency in this ticket. I would suggest that you create a new ticket for such a specific topic, if you still think that it is relevant.

FYI @MariMakes

closed 2022-11-02 by jellespijker

benjamin-9T, 2022-11-06:
The Problem for me was that C++ tool were not isntalled for Visual Studio 2022. Then the message that pops up is that Visual studio cannot be found