Ultimaker/Cura #12943: .3mf  files don't clearly offer load models only like it use to

Ser-filco, 2022-08-03:
### Application Version

5.1

### Platform

Win10

### Printer

S5

### Reproduction steps

drop .3mf file or open it

### Actual results

Screw up my unsaved printing profile 

### Expected results

Clear Msg
[Open 3D models Only] | [Open Models with Print settings]

I would imagine "models only" should be the default especially if the printer and/or material is not matching your current setup

Someone potentially could get his house burn down if someone send him a file with nozzle or bed temperature set to 1000°c  
... i sure hope you have some warnings built in for that ...

basically this window is missing:
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

all i got

GregValiant, 2022-08-03:
Thanks for the report.
Not all 3mf files are project files.  Some can just contain a model without printer or setting information.
If you open a model in Cura and then select "File | Export" the default file type is "*.3mf" and it will contain only the model.
If you use the "File | Save Project" command that will also create a *.3mf file but it would have your printer and settings in it.  If you open that file as a project it will definitely over-write your current settings because that is what it is designed to do.
When I open a project file I _always_ get the message box.
[image]

But when I open a 3mf model file there is no message box.  The model appears on the build plate in the same manner that an STL would.  There are no settings in the file to over-write my settings.

I think the developers would want to see a Cura log file.  The log file (Cura.log) can be found by going to "Help | Show Configuration Folder".  It's a text file and I believe you can post it here without zipping it.

lorenzoromagnoli, 2022-08-08:
As @GregValiant mentioned, when you open a Cura project file (.3mf) you are asked if you want to load just the model or the entire project. While I understand your case I can imagine that other people who regularly work with project files might have a different opinion regarding what should be the default. 

If you always to load model files and never be asked again to load the entire project please click "remember my choice" and you will never see that question again :)

closed 2022-08-08 by lorenzoromagnoli