Ultimaker/Cura #12944: Cura not launching due to Qt platform plugin error

ParryHotter95, 2022-08-03:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

None

### Reproduction steps

I tried to install both 5.0.0 and 5.1.0 versions of Cura under Windows 11, but it's still not starting due to the same error:
```
---------------------------
Ultimaker-Cura
---------------------------
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

---------------------------
OK   
---------------------------
```
Installing qt5 by `pip install pyqt5` and `pip install pyqt5-tools` didn't help.
I also tried running it as an Administrator.

### Actual results

It's not possible to start an application.

### Expected results

Program starts.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No log seems to be generated. Only files in AppData folder are related to older (4.x) versions of Cura.

jellespijker, 2022-08-03:
We can only help you if you provide us with the log and/or project files.

The log file, located at:
- Windows: `%APPDATA%\cura\<Cura version>\cura.log` or usually `C:\Users\<your_username>\AppData\Roaming\cura\<Cura version>\cura.log`
- MacOS: `$HOME/Library/Application Support/cura/<Cura version>/cura.log` (OSX)
- Linux: `$HOME/.local/share/cura/<Cura version>/cura.log`

Also be aware that Cura 5.+ Use PyQt6 not PyQt5. But that shouldn't matter to an end user like you.

ParryHotter95, 2022-08-04:
As I mentioned before, log file is not being generated. If you know how to force the program to generate any output before this exception occur, please instruct me. Project files are not relatable to this topic, as this is not project-related issue.

jellespijker, 2022-08-04:
I would suggest that you try uninstalling CURA and look into the program folder to see if you needs to do some manual cleaning up there. If you see a folder CURA 5.x delete it manually and then try reinstalling it.
Let me know if this will get you a bit further.

ParryHotter95, 2022-08-04:
I found a solution.
Until now, I tried installing it in my custom folder on D: drive. It's path was literally: `D:\Pliki programów`
After installing it in default `C:\Program Files` it worked.
Maybe it was the diacritic in letter "ó" causing the problem?
There should be solution for this to avoid people falling for this trap. Maybe path checking during installation will be helpful if problem lies in third party project (meaning pyQt) and can't be resolved directly.

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue and wondered if it's still something you are encountering. 

You can download our latest version here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-11-03 by github-actions[bot]