Ultimaker/Cura #12948: under extrusion in larg p

superfigo01, 2022-08-04:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender3pro

### Reproduction steps

I'm trying to print a piece that is pretty big, I've sliced it as usual  and the file looks good
[myissue.zip](https://github.com/Ultimaker/Cura/files/9258523/myissue.zip)

### Actual results

The printed model shows some stripes near the seam. Looking at it closer I see that it is an underextrusion problem. The flow rate is ok in the other parts of the print. I've been trying to adjust coasting, and the retraction extra prime amount, but that probably isn't the issue, because there are pretty big blobs on the seam. I noticed that, even with 0 retraction extra prime amount, The nozzle looks like it's purging some material on the seam before starting moving. I think it's a slicer issue
[image]
[image]

### Expected results

none of the above

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[myissue.zip](https://github.com/Ultimaker/Cura/files/9258548/myissue.zip)

Vandresc, 2022-08-04:
Hi @superfigo01, does it still occur when disabling the coasting?

superfigo01, 2022-08-04:
yes. I get pretty much no differences changing the coasting

Vandresc, 2022-08-05:
Not really something I can easily see in the gcode analyser, also the profile is using quite some custom settings, which makes it hard to pinpoint the issue. If the blobs are what causing an issue, I think you could change the seam location and see if that fixes it. I do not see it on the Ultimaker printers and unfortunately, I do not own a Creality. Is your printer also "freshly" tuned? I see a lot of under extrusion in the layers.

NotLordCalvert, 2022-08-05:
@superfigo01  for what it's worth, this appears to be related to the wall ordering and to retractions somehow. I had the exact same issue on an Ender 3 when print a cylinder in 5.0. After reducing the retraction distance, the issue moved from the outside to the inside of the cylinder. I then changed the wall order to "Inside to Outside" and it disappeared entirely.

superfigo01, 2022-08-05:
Thank y'all. I actually might have found out a way to solve the blobs problem. Apparently the newest versions of cura are conflicting a bit with the power outage restart feature available on marlin. If you turn this feature off you'll see the blobs disappearing. Kinda disappointing, but seems to work. I'm currently still testing this out, but hope it can help someone else

nikoraasu, 2022-10-22:
> Thank y'all. I actually might have found out a way to solve the blobs problem. Apparently the newest versions of cura are conflicting a bit with the power outage restart feature available on marlin. [...]

I experience the same issue with Marlin so while it might be tied to the Marlin setting that you mentioned, there probably also is something else causing this

ydnatag, 2023-12-13:
[image]

I think I have the same issue but a little worse. The same pattern is in several regions of the parts.

My current setup is:
* Cura 5.4
* Default ender-3 parameters
* Ender 3 Neo printer

The 3d printer works perfectly using Cura 4.X.

GregValiant, 2026-08-19:
The current Cura version is now 5.13.0.  Is this still a problem?  I don't see it on my prints.

closed 2026-09-02 by github-actions[bot]