Ultimaker/Cura #12949: strange cursor movement to another field when changing first number of infill percentage value

gurkenfolie, 2022-08-04:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Mega S

### Reproduction steps

Deleting the first number on the infill percentage leads to automatic jump the cursor to the next field

### Actual results

When I delete the first number of the infill percentage (e.g. "20.0 %" is set and i delete the "2") the cursor jumps to the next field making it impossible to type the new value without interacting with the mouse to bring the cursor in the correct field again.

Also after removing the first number and retyping the first number in the infill percentage field (as said, after interaction to move the cursor to the correct field again) it jumps directly to the next field without letting me type in a second number (here again i need to place the cursor manually in the correct field again.

This is really annoying.

Hope you can fix it.

### Expected results

Expected: It (the cursor) should stay in the same field

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

Vandresc, 2022-08-05:
Hi @gurkenfolie , thank you for your feedback, I will bring this issue with the rest of the team.

Vandresc, 2022-08-09:
The ticket is now on our backlog, devs see Cura-9537 for our internal reference.