Ultimaker/Cura #12950: Slicing Error

MikeOxawopper, 2022-08-04:
### Application Version

5,1,0

### Platform

Windows 10

### Printer

Ender 3v2

### Reproduction steps

1. opened cura
2. clicked the file icon
3. selected a model
4. opened the model

### Actual results

Model appears on build plate and an instant warning box appears.
UNABLE TO SLICE MODEL Unexpected Error.
consider reporting the bug

 

### Expected results

Endless slicer bar scrolling
[error part 1.zip](https://github.com/Ultimaker/Cura/files/9258955/error.part.1.zip)
Happens with every part i load, program doesnt crash.
printed this part yesterday using 4.1.3 and 5.0.0
[cura.log](https://github.com/Ultimaker/Cura/files/9259005/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9259011/cura.log)
[error part 1.zip](https://github.com/Ultimaker/Cura/files/9259012/error.part.1.zip)

MikeOxawopper, 2022-08-04:
OK,
Solved it!
5.1.0 won't run correctly inside the PROGRAM FILES directory?
uninstalled it, restarted computer
Reinstalled to the root directory C:/Ultimaker Cura 5.1.0
and now it works

GregValiant, 2022-08-05:
@MikeOxawopper Thanks for the update.
I'll leave this open as my installation works fine from the Program Files directory.

MariMakes, 2022-08-26:
@MikeOxawopper I'm happy to see you got unstuck.

I've renamed the ticket to correctly reflect your analysis of the problem.
I installed 5.1 straight into my program files folder to see if I could reproduce your issue.
[image]

But I cannot reproduce your findings.
Using that version I can still slice. 🤔 

I'll defer the issue for now, but I'll keep it open so people can share if they are experiencing something similar.

GregValiant, 2026-08-19:
This report has gotten stale and I'll close it.

closed 2026-08-19 by GregValiant