Ultimaker/Cura #12951: Initial Layer Material Pressure

x-stride, 2022-08-04:
### Is your feature request related to a problem?

It's always hard getting that initial layer down right.

When perfect, the nozzle is so close the extruder is struggling hard feeding it through.
Even opaque materials are almost see-through being this thin!

### Describe the solution you'd like

An extruder that first jumps down to a preset Z-height over the build plate, then:
 - gives pressure as feedback when extruding, possibly by monitoring motor current or by adding a sensor in it
 - Add Cura support for it
 - Add GCODE's etc

### Describe alternatives you've considered

Sound? Bad idea.
Look and feel, is as we do it now.
A Z-sensor is never enough...

### Affected users and/or printers

All

### Additional information & file uploads

## TPU SQUEEZE
[image]

GregValiant, 2022-08-04:
As you noted,, there may well be additional hardware and software required.  Additional hardware means additional I/O capability on the mainboard and software to deal with it.
If that were to be solved, I think the sensor would need to be right at the nozzle.  If it was anywhere else you would need to note the "compression load" of the filament itself.  Without a strain gauge on the filament I don't see that happening.  The flex in the bowden tube would make any reading meaningless.

My printer is an Ender 3 Pro and I level with a piece of parchment paper.  I noticed inconsistencies when a print would start.
I Auto-Home the printer and then level 4 corners and then start the print.  The G28 in the startup gcode causes the Z to home again changing all my careful leveling as the Z switch isn't very repeatable.  My fix was to change the G28 in the StartUp Gcode to G28 X Y.  Now after leveling the Z doesn't home again and so the print starts with the Z gap right where I set it.  It's exact every time.
Whether you level my hand, or use an ABL system, it's all about consistency.  If your layer height is .20 but your leveling is _always_ high by .04 then you will under-extrude.  If it is low by .04 then you will over-extrude and maybe block the nozzle opening.  Adding a Z-offset to compensate would get it right on every time provided you are consistent.
There is a Z-Offset plugin available in the MarketPlace.

x-stride, 2022-08-05:
Like you, also an Ender 3 Pro, with a direct drive extruder. Approximately 80 mm from the wheels pushing filament to the nozzle. z height matters, but also flow, temp and speed. I believe using extruder pressure can help a lot, also on increaeds surface quality and print accuracy.

closed 2022-09-21 by MariMakes