Ultimaker/Cura #12952: Additional margin of a brim is applied to all layers of a print

Triangulix, 2022-08-04:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Have a model that reaches over its footprint in at least one direction.
Place the model close to the border of the print surface so that its top gets within 10mm of the border while its bottom is far away from the border.
Enable a brim of 15mm.

### Actual results

The model can no more be sliced.
It seems that the 15mm additional margin for the brim gets applied to all laywers of the print, not only the first layer where the brim is.

### Expected results

The brim should only affect the first layer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothin to add

GregValiant, 2022-08-04:
Thanks for the report.
This is a duplicate of (or at least similar to)  #10209.
Please refer to @Ghostkeeper 's comments in that thread in regards to the "bounding box" of the model.
I will leave this as a bug report and add the duplicate label.

Triangulix, 2022-08-13:
After reading the issue you named as duplicate I hope that everyone understands that I am specifically talking about brim and raft. My issue is NOT that Cura prevents me from starting a print where the upper part of my object is hanging out of the printer.

GregValiant, 2022-08-13:
I added the duplicate because prior to slicing Cura only knows the bounding box of the model.  That includes any overhanging areas.  That is why this model won't slice in the position it is in.  That appears to be the crux of what is going on in #10209.
[image]

This is the model flipped around and pushed below the build plate.  The bounding box intrudes into the area reserved for the skirt/brim/raft and so this orientation won't slice either though it looks like it shouldn't be a problem.
[image]

I'll remove the Duplicate tag for you.

GregValiant, 2026-08-19:
I'll go ahead and close this as it is 4 years old and is inherent behavior in Cura.

closed 2026-08-19 by GregValiant

Triangulix, 2026-08-21:
Who cares about Cura anymore? It's 2026.

GregValiant, 2026-08-21:
Professionals who require flexibility.