Ultimaker/Cura #12953: Incorrect Print Time and Estimated Time Left - Jon Name Incorrect

blue68camaro-git, 2022-08-04:
### Application Version

5.1.0

### Platform

macOS Safari Raspberry PI

### Printer

Ender 3 Pro

### Reproduction steps

Brought STL file in and sliced. Sent to OctoPrint. Monitor Tab active and displays above incorrect information. 4.13.1 works fine.

### Actual results

Wrong information displayed on Monitor Screen. Time does change to less than 50% of original. Example: 30:18:22 for a 3:08:00 print job. then when printing started, 13:20:12. I have 2 Ender 3 Pros and setup identical, software, hardware, versions etc. One is using Cure 5.1.0 and the other 4.13.1. 

### Expected results

Correct information. Time is off by 20 hours more or less. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have nothing more to add.

Vandresc, 2022-08-05:
@fieldOfView, is this something related to your plugin?

blue68camaro-git, 2022-08-05:
I do not have this plugin installed.

MariMakes, 2022-08-26:
Hey @blue68camaro-git 

Thanks for your issue. 
As part of the Ultimaker team we focus on making sure that the monitor page works for Ultimaker printers.
Support for OctoPrint is contributed by our community members.
If you ask nicely they can help you with troubleshooting over at https://github.com/fieldOfView/Cura-OctoPrintPlugin

I'm closing this issue since it's not a Cura issue.

closed 2022-08-26 by MariMakes