Ultimaker/Cura #12954: filament cost is shared across all tpu filaments, regardless of vendor

doke2, 2022-08-04:
### Application Version

5.0.0

### Platform

Ubuntu 20.04 LTS

### Printer

TronXY Gemini S

### Reproduction steps

edit material Novamaker TPU to set price to $23.98
edit material TronXY TPU to set price to $26.99
go back and look at first TPU, Novamaker TPU, price has changed to $26.99
go back and look at generic TPU, price has also changed to $26.99.

### Actual results

Changing the price for any kind of TPU, regardless of vendor, changes the price for all TPU filaments.  

### Expected results

Each vendor-type-color of filament should be able to have it's own price.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info

Vandresc, 2022-08-05:
Hi @doke2 , I am not able to reproduce it, for me it keeps the given value. Could you please attach a project file so I can get access to the material? Pls zip the project file and add it via github. Thank you!

doke2, 2022-08-05:
Please close this issue as "user is an idiot".  I forgot to unlink the material profiles.  Sorry for wasting your time.  Thanks for answering anyway.

closed 2022-08-05 by doke2

Vandresc, 2022-08-05:
:) No worries, glad that it all worked out!