Ultimaker/Cura #12956: Concentric Support Infill Glitch

drknvdr, 2022-08-05:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Anycubic Mega X

### Reproduction steps

1. Sliced a model oriented at 45 degrees to the Y axis with concentric support infill.

### Actual results

[image]

### Expected results

Travel path for support infill showed deviations that lead to print failure. Pattern is erratic.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[AMX-CE3 STICK BUDDY 100%.zip](https://github.com/Ultimaker/Cura/files/9264895/AMX-CE3.STICK.BUDDY.100.zip)

GregValiant, 2022-08-05:
Thanks for the report.
That model is "not watertight" and there are internal errors.  Those errors can cause Cura to slice incorrectly and appears to be affecting the support infill.  This report is from [formware.co/stlrepair](https://formware.co/OnlineStlRepair)
--> 7 Non-manifold edges (?)
--> 14 Inverted faces (?)
--> 14 Duplicate faces (?)
-> Vertex count changed from 240200 to 241217 (+1017)
-> Triangle count changed from 480388 to 482630 (+2242)
MS 3D Builder was also able to repair the file.

Slicing the repaired file with your settings the support looks like this:
[image]
That view is a lot different from what you have.  I did not change any settings in the 3mf file, but I did need to repair the model and re-insert it in Cura.

What I do see here is that no matter what I change the "Support Density" setting to; it remains 0% in the actual slice.  This seems to be peculiar to the "Concentric" pattern.  Switching to "Lines" the support structure changes to this:
[image]

Which is the density I would have expected to see from "Concentric" as well.
A workaround for you would be to repair the model and maybe to change the Support Infill Pattern to something other than Concentric.

So I am unable to duplicate the "jig-jag" problem you are reporting in the Support Infill.  I do see a bug here with the infill density not affecting the slice.
The Cura Team will take a look at this.  My project file contains the repaired model.  This may be a duplicate of #12535 .

[GV_repaired_STICK BUDDY.zip](https://github.com/Ultimaker/Cura/files/9268634/GV_repaired_STICK.BUDDY.zip)

drknvdr, 2022-08-05:
Wow! Thank you for the detailed information. The problem was odd because
linear support infill sliced fine but concentric wouldn't. The only reason
I was using concentric was due to the model's organic shape. Though it
would offer adequate support without being a beast to remove. This model
also has printed fine in the past but I've now oriented it to hopefully be
able to print on an Ender 3 as well as a Mega X while trying to keep print
time around a day and under-ish 250g of filament. Still working out a
couple bugs to get it to print. Namely the rear points not sticking long
enough to be bridged into the rest of the model. As soon as I can nail this
then I should have a viable prototype that I can take to market. Thanks!

Jason

GregValiant, 2022-08-05:
You might have to move the model up a couple of mm's and allow support to build under those tips.  I would suggest enabling the support brim at maybe 4mm to make sure nothing breaks loose.

GregValiant, 2026-08-19:
This doesn't seem to be a problem in the current version (5.13.0) and I'll close this.

closed 2026-08-19 by GregValiant