Ultimaker/Cura #12958: Doesnt slice

Ironkiki3000, 2022-08-05:
### Application Version

5.0.1

### Platform

windows 10

### Printer

LK1

### Reproduction steps

the file size is too big 
doesnt slice 

### Actual results

crashed/ didnt slice

### Expected results

should be sliced 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[output_log.txt](https://github.com/Ultimaker/Cura/files/9266203/output_log.txt)

GregValiant, 2022-08-05:
I think a project file will be required here.
Load the model and use the "File | Save Project" command to create a 3mf project file.  Zip the file and post the zip here.
Without knowing anything about the model or how you are trying to slice it puts a halt to any sort of investigation.

Ironkiki3000, 2022-10-11:
Thank you , the file was too big for me to upload. I used an older version of cura to slice it . 
Thank you
SincerelyFarhan Mashroor 

Closed #12958 as completed.

GregValiant, 2026-08-18:
I'll close this.

closed 2026-08-18 by GregValiant