Ultimaker/Cura #12961: Cura 5.0 5.1 Dont show the button "Select settings" in "Per Model Settings" w/ 1366x768px resol

Tano2950, 2022-08-05:
### Application Version

5.0 5.1

### Platform

Windows 11

### Printer

Hellbot Magna 2 300

### Reproduction steps

1. My screen is HD 1366x768
2. I select a support blocker
3. Click on "Per Model Settings" 
4. The GUI don´t show me the button "select settings" 

### Actual results

The GUI don´t show me the button "select settings"
[image]
 

### Expected results

 The GUI should show me the button "select settings"  in the "Per model Settings"
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Vandresc, 2022-08-08:
Hi @Tano2950, we couldn't reproduce on our test machines with your given configuration, the per model setting window pops up higher which makes the button always visible. 
As a workaround you can add the settings you desire in the Normal model window (first icon) and then move to Infill mesh only.

GregValiant, 2026-08-18:
I'll close this.

closed 2026-08-18 by GregValiant