Ultimaker/Cura #12962: Different wall settings for top/bottom skin layers

traviselam, 2022-08-06:
### Is your feature request related to a problem?

Basically what I'm looking for is the ability to reduce the wall count on top/bottom layers for increased surface quality without sacrificing strength. I've ended up with inconsistent results with ironing enabled but got pretty good results with a skin line width of 0.28mm on a 0.4mm nozzle based on this video: https://youtu.be/jkAqTWmVRVA

The problem is if you increase the wall line count to something like 4 or 6 for strength, you sacrifice the area on the top of your print that can be adjusted by the skin settings.

### Describe the solution you'd like

I would like to have a different wall line count for top/bottom skins.

### Describe alternatives you've considered

I understand I could use a support blocker for this but the support blockers are not a feasible solution for complex models.

### Affected users and/or printers

All Cura users would benefit from this.

### Additional information & file uploads

[image]

Ideally in the picture above, I'd like only to have 1 or 2 walls in the skin layer.

MariMakes, 2022-08-22:
Hey @traviselam 

I discussed your challenge with the team. 
It is quite unclear what to do when you have both infill and skin in the same layer, what wall line count to choose in that situation?

A situation with a flat surface halfway through. 
[image]
[image]

Or a situation with a sloped surface like these. 
[image]

How would you expect the behavior to be here?

traviselam, 2022-08-23:
@MariMakes thank you very much for taking the time to consider my request! And you bring up some great scenarios that I hadn't thought of. I realize now that this is a more complicated problem than I thought. 

After looking at the models you suggested, my initial thought was that maybe you could use the same logic as your "ironing" setting. I imagine it would be something similar to "only top layer" vs "all layers". 

I'm obviously not familiar with the slicing engine. Still, if there was a way you could determine that a skin didn't have any layers directly above it and was "visible" then you could change the wall settings for that entire layer with the top visible skin. This would result in more surface area for the skin by reducing the wall count only for that layer. 

In addition, I'm sure it would be an expensive operation, but maybe you could have some sort of "minimum skin area" that would determine if the "skin wall count" would affect that layer.  This, I think, would solve the issue with minimal skin area regions in certain models. 

The sloped surface you posted is a tricky one, I'd be curious to see what the entire model looks like from the top down. And as far as the first image with a flat surface half way though, I'm wondering if you could detect that the skin area isn't in "contact" with any of the "outer walls" and leave the walls alone for that layer. I believe in that scenario, the "top surface skin line width" setting would work perfectly for the half way point and the top layer would be the only issue. 

I hope all that makes sense. Maybe this isn't something that is feasible but thank you for at least taking the time to investigate my feature. I'm relatively new to 3d printing, with only about 800 hours worth of experience, but I'm more than happy to answer any more questions you have and give the best input I can.

smartavionics, 2022-09-01:
See https://github.com/smartavionics/Cura/releases for a Cura variant that now supports single walls for the bottom skin and visible top skins.

[image]

klwemu, 2023-11-28:
If you have many walls the top layer looks ugly. A "skin" should be a "skin" and therefore include walls on the surface.. 
Check it out in Orca, they made it like that. Pretty staight foreward and it covers all mentioned special cases.

Regular wall
[image]

Skin setting
[image]

Setting
[image]