Ultimaker/Cura #12963: Slicing failed error

MrD-RC, 2022-08-06:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Prusa i3

### Reproduction steps

Imported my model. Rotated so that the flattest part is on the bed. Hit slice.

[image]

### Actual results

Slicing error message appears

### Expected results

The model should be sliced, ready for printing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura slicing error project.zip](https://github.com/Ultimaker/Cura/files/9274568/Cura.slicing.error.project.zip)

[cura.log](https://github.com/Ultimaker/Cura/files/9274570/cura.log)

GregValiant, 2022-08-06:
Thanks for the report and tanks for including the supporting files.
If you move the model to a different point on the build plate it should slice.
This may be a duplicate of #12932 which would also appear to involve the model location when the file is opened in Cura.
The Cura Team will take a look.

GregValiant, 2026-08-18:
I'm going to close this as the problem was addressed in the last couple of years.

closed 2026-08-18 by GregValiant