Ultimaker/Cura #12965: No support generated for layer that starts 1 layer above bed

ggee, 2022-08-06:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 3

### Reproduction steps

Import model https://www.thingiverse.com/thing:5447783/files
scale to 25%
rotate feet down
enable support and brim

### Actual results

The tip of the tail, which starts 1 layer above build plate is not getting any support or brim generated.  With no support, the tail fails print after a few layers since no bed contact.  I checked Cura 4 and support was generated properly.

### Expected results

Expected brim and support for tail.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Vaporeon_low.zip](https://github.com/Ultimaker/Cura/files/9275442/CE3_Vaporeon_low.zip)
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/9275444/cura.log)

GregValiant, 2022-08-06:
Thanks for the report.
Your settings indicate that the point of the tail _should not_ be supported.
When Cura is set up to provide support for a very small area with steep sides and at a proper top distance there is no problem.
Make some changes to your settings and let me know how it works out.  In the meantime I've removed the bug label as this appears to be a settings issue.
[image]

ggee, 2022-08-06:
Thanks.  It was that setting.

closed 2022-08-06 by ggee