Ultimaker/Cura #12968: Cura 5.1.0 Report Failure Slicing 2D picture

Lupis007, 2022-08-07:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender-3/Ender-3S/Ender-3Pro

### Reproduction steps

Trying to slice a picture to make a 3D image of it, but it reported that it failed. Have not had issues with other pictures prior so unsure what is going on with it. 

Attempted to close Cura and reopen it in case my other instances had caused an issue, but after all instances of Cura were closed and it reopened again, it still stated it failed. 

### Actual results

Reports the slicing failed without continuing, does not give any specific reason just asks for me to report the bug. 

### Expected results

Picture should slice and allow for me to print it. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Indyandcopperremake.zip](https://github.com/Ultimaker/Cura/files/9276036/CE3_Indyandcopperremake.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9276037/cura.log)

[image]

GregValiant, 2022-08-07:
Thanks for the report.
Here is the report from [Fomware Online STL repair](https://formware.co/OnlineStlRepair)
**--> 123964 Naked edges (?)
--> 11 Planar holes (?)
--> 3034 Non-planar holes (?)**
--> 0 Non-manifold edges (?)
**--> 1568 Inverted faces (?)**
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 181621ms ------
-> Vertex count changed from 323051 to 264085 (**-58966**)
-> Triangle count changed from 516072 to 528166 (**+12094**)

Just to be clear...Did Cura create the model from a photo you brought in?  I think that would be where any bug is.  That model has a lot of errors for being so small.  After repair it did slice, but when I tried to slice it before repairing it Cura crashed hard.
[image]

Lupis007, 2022-08-07:
Cura created the model. It never stated it saw any issues with how the model looked or anything like that.

GregValiant, 2022-08-07:
Thanks.
I have only sliced a couple of images myself and I didn't have any trouble.  Something happened when creating that one though.  Someone from the Cura Team will take a look.

Vandresc, 2022-08-08:
Devs, added as comment to already existing Cura-9380

rburema, 2022-08-19:
Tested the initial .zip the OP posted on an early alpha build of 5.2 -- slicing succeeds now. Closing this. If this (eventually, when it's out) still occurs in 5.2 beta or later, please open a new ticket (or ask us to reopen this one).

closed 2022-08-19 by rburema