Ultimaker/Cura #12969: Problem with concentric supports

VlamirDM, 2022-08-07:
### Application Version

5.0.0 end 5.1.0

### Platform

Windows 10

### Printer

Qidi x-max

### Reproduction steps

1. Loaded the gear model.
2. in the support section set the type of support concentricity
3 . when changing the % density of support, in reality the density does not change
4. slicing

### Actual results

[image]

Even when installing additional support separately, nothing changes. everything is bad too

### Expected results

[image]

cura v 4.16.0

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

fix this problem as soon as possible.

VlamirDM, 2022-08-07:
I like to use the new version with its new developments. but concentric supports, other types of supports cannot be replaced in my case

GregValiant, 2022-08-07:
Thanks for the report.
This same behavior occurs on my system.  This is a (near) duplicate of #12956 .  Someone from the Cura Team will take a look.

GregValiant, 2022-08-07:
Closed in favor of #12808

closed 2022-08-07 by GregValiant