Ultimaker/Cura #12973: Support layer height and line width

Ingam0us, 2022-08-07:
### Is your feature request related to a problem?

I‘m always frustrated when my supports don‘t print good. Just right now, a 19h print failed because the support had a hole and broke. 

### Describe the solution you'd like

I wish I could change the support layer heigth, just as the Infill layer height, and also the support line width, because if they are thicker, they are more stable. Also a support line multiplyer would be helpful, because the one singe „wall“ around is always wobbly, but if there is a second, it is more than 10times more stable. 

### Describe alternatives you've considered

my supports crash with any settings… 

### Affected users and/or printers

It affects me with both of my printers, the Anycubic Viper and the Sovol 04

### Additional information & file uploads

[image]

GregValiant, 2022-08-07:
All of those settings are already present in Cura.  Turn on the Custom settings.  To the right of the Setting Search box is an icon with three lines on it.  Click on that and set visibility to "All".
In the "Material" section is "Support Flow" and "Support Interface Flow".
In the "Quality" section is "Support Line Width"
In the "Support" section is "Support Infill Thickness", "Support Wall Line Count",
The Support Infill Thickness must be set to multiples of the regular layer thickness.  If you set it to 2X layer height then the Support Infill will go down every other layer and at twice the flow rate.

MariMakes, 2022-08-29:
Hey @Ingam0us 

Welcome to Github 🚀 
I'm sorry your print job failed, that's super frustrating. 

@GregValiant is right. We have those and many other settings available.
You can learn more about them in the Settings Guide Plugin 
Did you get a chance to try these settings? 
Can you let us know if they worked for you? 
Then I can close the ticket. 😄

closed 2022-09-13 by github-actions[bot]