Ultimaker/Cura #12975: slicing failed with an unexpected error

Delphi-FPC-Lazarus, 2022-08-07:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Longer LK5 Po

### Reproduction steps

slice
I've uploaded files below

### Actual results

slicing failed with an unexpected error

### Expected results

slicing should work like always did before?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I've uploaded project file here:
https://www.file-upload.net/download-14981341/LLK5PRO_Amy_negativ.3mf.html
and stl file here:
https://www.file-upload.net/download-14981343/Amy_negativ.stl.html

slicing failed with an unexpected error
[image]

GregValiant, 2022-08-07:
My anti-virus didn't like that download site at all.  You will need to zip the files and post them here.
You might want to try opening that model in MS 3D Builder which should be included with Windows 10.  Open and then "Insert" the model.  I believe you will get a warning that it needs repair.

Delphi-FPC-Lazarus, 2022-08-07:
[Amy_negativ..zip](https://github.com/Ultimaker/Cura/files/9277806/Amy_negativ.zip)

Uploaded here. (as zip, containing stl and 3mf project file)
i've created those files myself (converted from image), so they're clean. just slicer seems to run in an issue slicing it.

Delphi-FPC-Lazarus, 2022-08-07:
btw. : no warning from 3d builder (viewer in windows 10 ) or cura

GregValiant, 2022-08-08:
When I open and then Insert in 3D Builder I got the warning...

[image]

And here is the report from Formware:
--> 1688 Naked edges (?)
--> 523 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 138 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 169183ms ------
-> Vertex count changed from 133098 to 120260 (-12838)
-> Triangle count changed from 266222 to 241136 (-25086)
It appears that errors are being made when the model is being created from the photo, but that's just my impression.

This looks like a duplicate of #12968 and the behavior has been noted by the developers.  It appears they have a ticket for this on their backlog.  I'll mark this as a duplicate and leave it up to them as to whether to close it or not.

Delphi-FPC-Lazarus, 2022-08-08:
oh, ok, thanks
error didn't disappear afer repair and export to stl but seems to disappear after repair and store as 3mf.

I'll keep on working on it ;)

GregValiant, 2026-08-18:
I'll close this.  I didn't have any trouble slicing the model (as is) using Cura 5.13.0.

closed 2026-08-18 by GregValiant