Ultimaker/Cura #12977: Concentric initial layer is not printed sequentially

tangcla, 2022-08-08:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Creality CR-6 SE

### Reproduction steps

I sliced a circular model with concentric initial layer.

### Actual results

Initial layers are drawn in rings at seemingly random intervals - it would do several walls, then move in (or out) and do some more, and then come back to fill the gaps

Uploading Untitled video - Made with Clipchamp.mp4…

.

### Expected results

I expected that the nozzle would move from inside to outside (or vice versa) to minimise print head movement and retraction.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've tried the same setting on a variety of different shapes (circle, square, and other odd shapes) but the pattern seems to still happen.

Vandresc, 2022-08-15:
Hi @tangcla , thank you for your feedback, I will bring this up for discussion with the rest of the team.

Vandresc, 2022-08-16:
Devs, see CURA-9551 for our internal reference. 

@tangcla we have a ticket on our backlog, thank you!

elmo40, 2022-11-19:
v5.2.1 still has this weird issue.
View photo:
[https://unsee.cc/album#FTrDHO2HFbSM4eQa](url)

jellespijker, 2022-11-22:
@elmo40 that is indeed the case since   the current fixes are for 5.3

Although there are still some edge cases we're working on, I do expect that those will be resolved for the 5.3 release

closed 2022-12-01 by jellespijker