Ultimaker/Cura #12979: Loss of the first layer when the part is moved under the table.

mrMastor, 2022-08-08:
### Application Version

5.0.1

### Platform

Windows 11

### Printer

FBG5

### Reproduction steps

1) Lower the workpiece below the level of the table.
2) Slice.
3) Check how to lay down the first layer.

### Actual results

First layer in place.

### Expected results

The first layer is not printed, the detail is in the air.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[1_not_Ok.zip](https://github.com/Ultimaker/Cura/files/9279816/1_not_Ok.zip)

GregValiant, 2022-08-08:
Thanks for the report
This appears to be a function of the slicing tolerance.
In the Experimental section:  If you set Slicing Tolerance to "Middle" then the empty first layers are removed with only the pin at Z=-0.3 and the other models at Z=0.
[image]

GregValiant, 2026-08-18:
I'm going to close this as stale.  There have been no further reports of this behavior in the 4 years since the report was made.

closed 2026-08-18 by GregValiant