Ultimaker/Cura #12984: ERROR SLICING

WWGG12, 2022-08-08:
### Application Version

5.1.0

### Platform

Mac OSX

### Printer

Creality CR X

### Reproduction steps

1. Imported STL
2. Tried to slice it.

I have sliced this 100 times in previous version of Cur
[Homepot mini v2 kopie.stl.zip](https://github.com/Ultimaker/Cura/files/9284756/Homepot.mini.v2.kopie.stl.zip)
a

### Actual results

ERROR SLICIN
[image]
G

### Expected results

no error ;-)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CCRX_Homepot mini v2 kopie.3mf.zip](https://github.com/Ultimaker/Cura/files/9284777/CCRX_Homepot.mini.v2.kopie.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9284799/cura.log)

GregValiant, 2022-08-08:
Thanks for the report.
I checked the model and can't find any errors but it won't slice.  This is not the first instance of this behavior but it is the first one that I couldn't get to slice at all.
I'll mark this as a duplciate and leave it open.  Someone from the Cura team will take a look at it.

WWGG12, 2022-10-08:
In 5.2.0 it will slice on a 100% scale. As soon as I increase the size of the model to 200 or more % the slicers fails

GregValiant, 2022-10-09:
I tried again and I can't get it to slice at any scale in 5.2b

emmav28, 2023-05-06:
i have the exact same issue is there any way to solve the problem

GregValiant, 2023-05-06:
I tried this again with 5.3.1 and it sliced at 100% scale but not when I scaled it up.  It sliced in Cura 4.13.1 at any 100% and at 200% scale.  4.13.1 is available [HERE](https://github.com/Ultimaker/Cura/releases?page=2).

emmav28, 2023-05-06:
Thank you I got 4.13.1and you are right it work great what I don't
understand is to why the release software and don't patch bugs

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.

I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker