Ultimaker/Cura #12986: Does not create support under Normal, only with Tree

pauliaus, 2022-08-09:
### Application Version

5.1

### Platform

win 10

### Printer

2+

### Reproduction steps

Slice with Normal support - none, slice with Tree - ok. 

### Actual results

Drooping overhang, failed print

### Expected results

Support generated

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[UM2_4x6_95r_18650_tray (3).zip](https://github.com/Ultimaker/Cura/files/9288253/UM2_4x6_95r_18650_tray.3.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9288264/cura.log)

GregValiant, 2022-08-09:
Thanks for the report.
Your "Minimum Support Area" is set to 4.0mm and the tips of the squares are smaller than that.  As a consequence they aren't being supported because you asked Cura not to support them.
Your "Support Z Distance" is 0.1 and your Layer Height is 0.25.  Cura will not change layer height for the top of the support.
Your "Support X/Y distance" is 1.0.  The tips of the squares that you want to support are small.  A generous distance like that will make for a very small footprint for the support and is a failure point.
I would suggest that you:
Set the "Support X/Y distance" to 0.6
Set the "Minimum Support Area" to 1.0
Set the "Support Z Distance" to either "0" or an exact multiple of your layer height.
Enable "Support Brim".

Given your settings, what I see here is "expected behavior" rather than a bug.  I'll leave it open so you can respond but I'll remove the bug label (at least for now).

GregValiant, 2022-08-09:
Ok.  It's back to a bug.

Vandresc, 2022-08-11:
The ticket will be brought up for discussion with the rest of the team.

Vandresc, 2022-08-12:
We created a ticket on our backlog, we are planning some fixes for the support feature soon. Devs, see CURA-9548 for our internal reference.