Ultimaker/Cura #12990: Dual extrusion slicer failure

Arnold-Muralt, 2022-08-09:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

ultimaker S5

### Reproduction steps

Insert files to print
select nylon + PVA materials
have per model settings enabled for generate supports 
Tell slicer to generate support with extruder 2 (PVA)
Slice!
Unknown error

### Actual results

Slicing Fails with an unexpected error, 
please consider reporting a bug

### Expected results

all parts should have support if selected in per model settings, slice should complete correctly

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Curalog and project file.zip](https://github.com/Ultimaker/Cura/files/9293210/Curalog.and.project.file.zip)

Arnold-Muralt, 2022-08-09:
I did some additional testing and it seems that tree mode is seemingly unaffected, something must be going wrong with the standard support calculations

GregValiant, 2022-08-09:
Thanks for the report.
All three models required repairs.  I used MS 3D Builder.
Bringing the repaired models into Cura they sliced and regular support was generated for the two models that have their Per Model settings set to "generate support".  The third model (the large one) did not have "generate support" enabled and so it was not supported.  Since the regular settings "generate support" is disabled that would be expected behavior.
The large model would not slice if Generate Support was enabled for Everywhere.
[image]

GregValiant, 2026-08-18:
This report has gotten stale and I'll close it.  The project sliced as it should in Cura 5.13.0.

closed 2026-08-18 by GregValiant