Ultimaker/Cura #12991: Slicing error

IJMFDavis, 2022-08-09:
### Application Version

5.1.0

### Platform

windows 11

### Printer

Creality CR-6 SE

### Reproduction steps

I tried to print CR6SE_Christmas_Ball_Lattice_with_Small_Twist(1) from thingyverse It then came up with a slicing error. Tried it in Creality slicer 4.2.8 and it took ages for it to slice but it seemed to work. Could be the model. I was just bored and wanted something to print. 
[Christmas_Ball_Lattice_with_Small_Twist(1).zip](https://github.com/Ultimaker/Cura/files/9294256/Christmas_Ball_Lattice_with_Small_Twist.1.zip)
e

### Actual results

Couldn't print with Cura.

### Expected results

For it to print

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9294307/cura.log)

GregValiant, 2022-08-10:
Thanks for the report.
There are errors in the model.  I was able to repair it using MS 3D Builder.  
I'll leave this as a bug report but it does appear that the model has problems.
Here is the repaired file.  Let us know if you can slice this one.
[Repaired_Christmas_Ball.zip](https://github.com/Ultimaker/Cura/files/9296178/Repaired_Christmas_Ball.zip)

[image]

IJMFDavis, 2022-08-10:
Hi Greg, 

Thanks for coming back so quickly. To be honest I wasn’t expecting a reply. 

Thank you for fixing the model. I’ll give it a go and let you know. 

Stay chilled! 

Many Thanks

Ian Davis

IJMFDavis, 2022-08-14:
Hi Greg,                  Bad news sadly. I tried to slice the model again and it still won’t work. Same error.  Any Ideas?  Many Thanks Ian Davis07960 877 220 From: Ian DavisSent: 10 August 2022 20:31To: Ultimaker/CuraCc: Ultimaker/Cura; AuthorSubject: Re: [Ultimaker/Cura] Slicing error (Issue #12991) Hi Greg,  Thanks for coming back so quickly. To be honest I wasn’t expecting a reply.  Thank you for fixing the model. I’ll give it a go and let you know.  Stay chilled! Many Thanks Ian Davis

GregValiant, 2022-08-14:
I'm not having a problem with it.
Here are two gcode files to give you something to do.  One is with supports on the inside and the other is "touching build plate".
I don't know what your preferred temperatures are.  You should open the gcode files in a text editor and check the M104, M109, M140 and M190 lines at the beginning.  They set the temperatures.
[2-Christmas_Ball.zip](https://github.com/Ultimaker/Cura/files/9334158/2-Christmas_Ball.zip)

IJMFDavis, 2022-08-15:
Hey Greg,

Wow, thank you so much. I really didn’t expect any of this. It’s like going back 40 years and dealing with someone with good customer service who actually cares. 

That worked a treat. It’s printing now. Let’s see what happens. 

Thanks again though, you’re a star.

Have a good week. 

Many Thanks

Ian Davis

GregValiant, 2022-08-15:
As Red Green always said..."Men, we're all in this together."
Of course he was talking about marriage and not 3D printing, but it seems appropriate.

GregValiant, 2026-08-18:
Time to close this one.

closed 2026-08-18 by GregValiant