Ultimaker/Cura #12993: Request that Coasting values follow Layer Height

Ergonomicmike, 2022-08-10:
### Is your feature request related to a problem?

Yes, in that Coasting values are not automagically corrected when Layer Height changes. This can cause gaps or 'pile ups,' depending on which way Layer height changed from a previous setting.

### Describe the solution you'd like

In the Ultimaker Manual, it says that 
Coasting Volume = Nozzle width * Layer height * length [of a cetain path].
https://support.ultimaker.com/hc/en-us/articles/360012614559-Experimental-settings

(This in contrast to the Cura Settings Guide, which says that Coasting Volume is a fixed value, generally equal to the cube of the nozzle diameter. )

If the former is correct, then when a User changes the Layer Height in Cura, the Coasting Volume should also be changed.

If Cura can't make this change automagically (which I'm guessing it can not), then I suggest that the Coasting Volume field, AND the Minimal Volume Before Coasting change to the color Yellow, to indicate to the User that these values need attention.

After the User sets new values, or confirms existing values, the Yellow should clear.

### Describe alternatives you've considered

Cura could make these two values automatic, with only the length of Coasting Volume and the length of Minimal Volume Before Coasting be user settable. That is, "How long a length do you think it will take the nozzle to coast?" and "How long a length do you want before coasting begins?"

This would make more sense to Users than volumes in mm^3, since the Preview shows the effect of Coasting in lengths of the paths.

### Affected users and/or printers

Everyone who uses Coasting.

Ergonomicmike, 2022-08-17:
I found a possible bug in Cura regarding coasting values not recomputing after the layer height change after initial height. I made a short video to show the devs.

https://jmp.sh/h3SNgQ1

MariMakes, 2022-08-22:
Hey @Ergonomicmike thanks for reporting and the nice video.

I've discussed the issue with the team and they are open to changing the formula of the Coasting Volume.
(It's currently a fixed setting)

It's a setting that is not commonly used in Ultimaker Printers so I'm curious...
What formula would fit your needs instead?

Ergonomicmike, 2022-08-23:
Well, thanks for asking me. But I'm a total newbie.

As a first try, I would go with the Ultimaker guys and use their Coasting Volume = Nozzle width * Layer height * length [of a certain path]. Let us chose how much of a gap we expect. Cura does the math, knowing the nozzle width and current layer height to calculate when to stop extruding.

The only reason that I found Coasting in Cura was because, if you've ever watched a professional painter paint a car, they always release the spray trigger just before coming to an end of a stoke so that the paint doesn't 'bunch up' at the end.

Kind of like a poor man's Linear Advance I suppose. And I had noticed bunching on the first layers of my prints (where it's most easily seen). So that's why I use it.

For me more of a purist thing.

And, of course, it's more ergonomic for users to think in terms of length of coast as opposed to coasting volume.

Ergonomicmike, 2022-08-23:
I should add that Coasting seems to help reduce Z-seam problems too, since the Extruder stops extruding just before a change in Z.