Ultimaker/Cura #13000: not posible add material

Zedral1, 2022-08-10:
### Application Version

5.1.0

### Platform

Windovs10

### Printer

Ender3

### Reproduction steps

If i add new material, it is no posible 
Iwould like add PLA, ASA, ABS and PETG it is no posible 

### Actual results

If i start cura software i dont  add material 
it is still freze on ultimaker Red Translucent PET
[image]
G

### Expected results

how to add material

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9302275/cura.log)

Vandresc, 2022-08-15:
Hi @Zedral1, could you please provide a project file as well? I need your configuration in order to reproduce the issue. Thank you!

MariMakes, 2022-08-26:
Hey @Zedral1,

How did you add your 3D printer?
I believe if you add a Creality Ender 3 from the Add Printer list, instead as a custom FFF printer you get a bunch of community-contributed materials. It already has the profiles for PLA, ASA, ABS, and PETG. 
[image]|
[image]

We also have a support page about custom materials 
https://support.ultimaker.com/hc/en-us/articles/360012565000-How-to-add-a-custom-material-in-Ultimaker-Cura-

Can you let us know if this solves your problem? 👼

closed 2022-08-29 by github-actions[bot]