Ultimaker/Cura #13009: Silicing fails periodically

Yurielanda, 2022-08-12:
### Application Version

5.1.0 beta +304

### Platform

Windows 10

### Printer

Creality ender 3 pro

### Reproduction steps

1. Selected files to slice
2. Clicked the slice button

### Actual results

Slicing fails

### Expected results

Slicing succeeds

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Crystal_Medium.zip](https://github.com/Ultimaker/Cura/files/9313127/Crystal_Medium.zip)

r2dane2, 2022-08-12:
Same issue as above, on the following:

Version: 5.1.0
Ender 3 Pro
Windows 11 64-bit

[CE3PRO_Front.zip](https://github.com/Ultimaker/Cura/files/9313288/CE3PRO_Front.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9313290/cura.log)
[image]

MariMakes, 2022-08-26:
Hey @Yurielanda,

I'm having some trouble reproducing your issue on Cura 5.1 Stable. 
Your files slices for me, did you try Cura 5.1 Stable?
If you do do you mind sharing your logs with us? 
[This is how you share them ](https://support.ultimaker.com/hc/en-us/articles/360016538000-Saving-Ultimaker-Cura-logs) 
[image]

@r2dane2 I took a look at your files. I see an engine crash in the logs that I can have someone from the team look at. But the .3mf file that you shared seems to be corrupted. Do you mind sharing it again?

closed 2022-09-09 by github-actions[bot]