Ultimaker/Cura #13012: Slicing Failed

legoboy177, 2022-08-12:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender3 V1

### Reproduction steps

Loaded File, Hit Slice
[Mid Rear-2.zip](https://github.com/Ultimaker/Cura/files/9315711/Mid.Rear-2.zip)

### Actual results

Error message appears saying it could not slice

### Expected results

Sliced file

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

N/A

GregValiant, 2022-08-12:
Thanks for the report.
This is a duplicate of several other reports.  I will mark it and someone from the Cura team will take a look.
I did get the model to slice by moving it left and forward on the build plate just a little bit and then rotating the model 15°.
[image]

GregValiant, 2026-08-18:
This should have been closed with the others.  I'll close it now.

closed 2026-08-18 by GregValiant