Ultimaker/Cura #13013: "Running from Source|" installation error

SWDontheway, 2022-08-12:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker

### Reproduction steps

1. after running 
`conan install . cura/5.1.0-beta@ultimaker/stable --build=missing --update -o cura:devtools=True -g VirtualPythonEnv`

### Actual results

cpython/3.10.4: WARN: Build folder is dirty, removing it: C:\Users\fay0\.conan\data\cpython\3.10.4\_\_\build\1e1dce5c5f980cc9662f9422f167ceffc2aa5e4e
cpython/3.10.4: Copying sources to build folder
cpython/3.10.4: Building your package in C:\Users\fay0\.conan\data\cpython\3.10.4\_\_\build\1e1dce5c5f980cc9662f9422f167ceffc2aa5e4e
cpython/3.10.4: Generator 'MSBuildDeps' calling 'generate()'
cpython/3.10.4: ERROR: Traceback (most recent call last):
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\site-packages\conans\client\generators\__init__.py", line 181, in write_generators
    generator.generate()
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 110, in generate
    generator_files = self._content()
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 333, in _content
    result.update(self._package_props_files(dep, build=False))
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 303, in _package_props_files
    result[comp_filename] = self._dep_props_file(full_comp_name, comp_filename,
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 229, in _dep_props_file
    dom = minidom.parseString(content_multi)
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\xml\dom\minidom.py", line 1998, in parseString
    return expatbuilder.parseString(string)
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\xml\dom\expatbuilder.py", line 925, in parseString
    return builder.parseString(string)
  File "C:\Users\fay0\AppData\Local\Programs\Python\Python39\lib\xml\dom\expatbuilder.py", line 223, in parseString
    parser.Parse(string, True)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 6, column 33

ERROR: Error in generator 'MSBuildDeps': not well-formed (invalid token): line 6, column 33

### Expected results

successful installation without error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

SWDontheway, 2022-08-12:
solution
[](https://github.com/conan-io/conan/issues/11822)

SWDontheway, 2022-08-12:
solution
https://github.com/conan-io/conan/issues/11822

closed 2022-08-12 by SWDontheway

jellespijker, 2022-08-12:
Yeah we also run into that one. Make sure you skip conan 1.51.x

ZacharyHowell-Essentium, 2022-08-19:
@jellespijker 
Would you mind elaborating on how one can skip conan 1.51.x?
Thanks!

jellespijker, 2022-08-19:
`pip install conan==1.50.1 --force` should do the trick