Ultimaker/Cura #13015: Request for "Linear Advance" type feature for fan speed when bridging

Ergonomicmike, 2022-08-12:
### Is your feature request related to a problem?

Yes, because fan speed changes are not instantaneous.

### Describe the solution you'd like

I'm experimenting with the Bridging settings. Fan speed can be changed when printing bridges. It would be nice if fan speed would begin to increase shortly before bridging begins. (I suppose a user setting for advance time, in seconds?)

### Describe alternatives you've considered

None.

### Affected users and/or printers

All those printers with a parts' fan(s).

smartavionics, 2022-08-12:
See https://github.com/smartavionics/Cura/releases for a Cura variant that provides this feature. It has a Cooling Fan Lag setting that lets you specify how long it takes for the fan to spool up/down.

Ergonomicmike, 2022-08-12:
Tnx. As a former Glasair pilot, I am always interested in smart avionics.

MariMakes, 2022-08-22:
Hey @Ergonomicmike 
I brought your issue up to the team and they are interested in the potential. 
I created a ticket on the backlog. For internal reference: CURA-9554

Ergonomicmike, 2022-08-23:
Thanks! I don't think it needs to be fancy, like ramping up to speed in a Bezier curve sort of way. Just a simple number of seconds before the fan comes up to its bridging speed. 

Most of us print slowly (50 mm/sec) and so we have a feel for how long it will take the head to move before coming to a bridge. For example, a 1 second advance would be better than none. I would probably be happy with 1 or 2 seconds. (It's not like my fans have tons of inertia. They come up to speed rapidly.)