Ultimaker/Cura #13017: Acceleration Tower Slice Error

Fleazus, 2022-08-12:
### Application Version

5.1

### Platform

Windows 10

### Printer

Ender 6

### Reproduction steps

Added Calibration from extension, Acceleration Tower.  Attempt to slice.

### Actual results

Failure to Slice

### Expected results

Successful Slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Acceleration Tower.zip](https://github.com/Ultimaker/Cura/files/9330276/Acceleration.Tower.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9330286/cura.log)

GregValiant, 2022-08-13:
Thanks for the report.
The Acceleration Tower is from the Calibration Shapes plugin by @5axes .  This error has been noted in #12932 and #12876.  I'll mark this as a duplicate.  I'll leave it open so 5Axes sees it.  It was reported on his Github page as well.

GregValiant, 2026-08-18:
I'm going to close this as the plugin has been taken over by another person.

closed 2026-08-18 by GregValiant