Ultimaker/Cura #13018: Separate speed for first layer above infill

Triangulix, 2022-08-12:
### Is your feature request related to a problem?

I'm currently optimizing a profile for speed and I noticed that bottom layers and generally layers on top of a flat surface can be printed much faster than layers above infill.
The infill rips the printed line apart when the nozzle moves over it and the material curls up and creates bumps which ruin everything printed on top of this layer.

### Describe the solution you'd like

To mitigate this I would like to print the first top layer above infill at slower speed than the rest of the top layers.

### Describe alternatives you've considered

I could lower the Top/Bottom speed value but this costs a lot of time in large objects and is not necessary for all other layers except the one above the infill.

### Affected users and/or printers

Me and probably lots of others.

smartavionics, 2022-08-13:
See https://github.com/smartavionics/Cura/releases for a Cura variant that kinda provides this feature. It has a setting called Bridge Sparse Infill Max Density that lets you print skin over sparse infill using the bridge skin settings rather than the normal skin settings....

[image]

Triangulix, 2022-08-13:
> See https://github.com/smartavionics/Cura/releases for a Cura variant that kinda provides this feature. It has a setting called Bridge Sparse Infill Max Density that lets you print skin over sparse infill using the bridge skin settings rather than the normal skin settings....
> 
> [image]

That looks like a solution. Do you plan to continue maintaining this fork so that it keeps up with Cura's development?

Vandresc, 2022-08-15:
Wouldn't Top Surface Skin Speed cover your request? Or there is only the layer on top of the infill that you want to decrease the speed for? Will bring it up for discussion with the rest of the team.

Triangulix, 2022-08-15:
Hi @Vandresc, I could archieve half of the desired result using "Top Surface Skin Layers" with a "Top Surface Skin Speed" greater than the "Top/Bottom Speed" but on one hand I had to manually determine the number of Top Surface Skin Layers, depending on the desired "Top Thinkness" and the resulting number of layers and on the other hand I still had to set the bottom layers to an unnecessarily low speed.

MariMakes, 2022-08-22:
Hey @Triangulix 

I brought it up to the team and they mentioned that the setting Bridge Skin Support Threshold.
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/bridge_skin_support_threshold.md

Triangulix, 2022-08-22:
Hi @MariMakes 
I didn't know that this feature was alread available in Cura 5.1.

The option "Bridge Skin Support Threshold" seems to have no effect on skin that is printed over infill-filled areas but the option "Bridge Sparse Infill Max Density" does.

This could be a solution for my problem but I have to test that. It is important that the skin layers still fuse to the infill and do not become a bridge that just happens to be over the infill.

Triangulix, 2022-08-23:
The bridging feature solves the problem.

MariMakes, 2022-08-23:
Awesome! 🎉 
That's great to hear, thanks for updating us. 
I'll close the issue.

closed 2022-08-23 by MariMakes