Ultimaker/Cura #13019: "Skin Overlap Percentage" not used for skin if "Extra Skin Wall Count" is > 0

Triangulix, 2022-08-13:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Set "Extra Skin Wall Count" to a value > 0 and set "Skin Overlap Percentage" to a value you can easily see in the preview, like 60%.

### Actual results

The overlap which should help to attach the ends of the skin lines to the walls is not applied to the skin lines but to the additional wall line. 

### Expected results

I know that someone will claim that the behavior is correct since the additional wall line is technically part of the infill.
But this doesn't change the fact that the overlap only helps if it is applied where ends of lines need to stick to walls.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2022-08-13:
Thanks for the report.
I am showing the same behavior in Cura and reading the gcode files into AutoCad confirms the problem.
The developers will take a look at this.

Piezoid, 2022-08-13:
also, bridge settings are not applied to skin extra walls either.

Vandresc, 2022-08-16:
We have a ticket on our backlog, CURA-9381. I will link the git ticket to ours. I will close this as a duplicate of https://github.com/Ultimaker/Cura/issues/12294, please feel free to follow any progress in the linked ticket. 
Thank you for your report!

closed 2022-08-16 by Vandresc

PhilBaz, 2022-10-01:
@Vandresc 

This has not been fixed in 5.2 beta....

Also, please, bridge and surface-skin each need their own skin overlap % settings.