Ultimaker/Cura #13020: TSP Infill

Triangulix, 2022-08-13:
### Is your feature request related to a problem?

Most infill patterns have crossing lines. These crossing lines, when printed at high speed, can create little bumps at every crossing. This creates vibration in the print head moving over these bumps, leading to decreased print quality, mechanical stress and even failed prints because of the bumps adding up layer by layer.

### Describe the solution you'd like

To solve this issue we need an infill pattern that consists of a single uninterrupted line which is following a maze like pattern with enough turns into every direction to make it sturdy. To make it really sturdy, instead of crossing each other the "lines" could touch each other regularly without crossing.

There are generators for this kind of pattern available on the internet. Have a look at these websites:
https://www.vidarholen.net/contents/blog/?p=1062
https://www.autoitscript.com/forum/topic/190921-tsp-art-generator/

Or google for "TSP art" with image search.

### Describe alternatives you've considered

The patterns "Cross" and "Concentric" are without crossing lines but both are lacking the direction changes necessary to give the pattern stability.

### Affected users and/or printers

Cura users

### Additional information & file uploads

[image]
[image]
[image]

Vandresc, 2022-08-15:
Hi @Triangulix, did you try enabling the Connect Infill Polygon? That together with cross or lightning (even Lines), can generate an infill pattern that consists of a single uninterrupted line/maze. Make sure Connect Infill lines setting is enabled as well.

Triangulix, 2022-08-15:
Hi @Vandresc, the pattern "Cross" unfortunately is bad at supporting. At a 45° angle, which is also what the layer on top of the support will use, it doesn't support at all avery few lines.
[image]
Also there is not way to get the line of the Cross pattern to touch from time to time to give it rigidity.

I tried the option "Connect Infill Polygons" and I didn't see any difference. I also didn't expect any difference since this option in my understanding merges the infill patterns of overlapping volumes.

Maybe you meant "Connect Infill Lines"? This option is very useful for better connecting the infill to the walls but it does not prevent crossing of lines which is my the main point.

The pattern "Lines" doesn't cross lines but it also provides barely any stability since the layers only touch at the crossing points of the lines.

When the pattern "Lightning" is selected the options "Connect Infill Lines" and "Connect Infill Polygons"are not available.

Triangulix, 2022-08-15:
I just had another idea of a pattern that prevents crossing lines but offers rigidity by touching lines
[image]
.
The idea is to go around clockwise or counter-clockwise and place arcs along the outline. When the round is finished, move one step in and repeat placing arcs around the outline created by the outer ring of arcs.
Of course the arcs should be nicer and more regular than mine. I cannot draw. Especially not with a mouse.

Triangulix, 2022-08-15:
Clarification: When I write about getting rigidity by touching instead of crossing lines I mean something like this:
[image]
:

MariMakes, 2022-08-16:
Hey @Triangulix 

@BagelOrb made a cool pull request that I think fits your needs!
You can find it here: https://github.com/Ultimaker/CuraEngine/pull/1356

If you are able to turn your pictures into surfaces, you can use them to generate your own infill.
Here is a .3mf with an example of how it could work with Honeycomb infill:
[full_hex_infill.zip](https://github.com/Ultimaker/Cura/files/9349594/full_hex_infill.zip)

You can also use it to make reverse gradual infills like this:
https://user-images.githubusercontent.com/40423138/184850511-854610af-56e8-409f-8d7e-5be737bbcf98.mp4

Can you let me know if this helps?

Triangulix, 2022-08-16:
@MariMakes How can a modifier mesh that can only set parameters give me a totally new infill pattern?

MariMakes, 2022-08-16:
Hey @Triangulix,

If you change the following settings you can have any model be your infill.
1. Set your infill to 0%
2. Load a model
3. Select Model
4. Define Modify Mesh Type to: Modify Settings for Overlap
5. Select Settings
6. Select Surface Mode
7. Define Surface Mode as Surface
8. Slice
[image]

It get's even better! You can load pngs in Cura so I used one of your attached pictures to define the infill for my model as an example.

I used these settings to import the png. 
[image]

I made it overlap with my model and I got this result
[image]
[image]

Can you try it and let me know if it works for you?

Triangulix, 2022-08-20:
This looks really nice. But to get such infill patterns for objects of all sizes and shapes I assume we would need some kind of pattern generator.

Triangulix, 2022-08-22:
I just realized another beneft of TSP style infill patterns:
Since the infill is one continuous line, in many situations the whole layer could be printed without a single travel move.