Ultimaker/Cura #13021: Softer transition between wall lines

Triangulix, 2022-08-13:
### Is your feature request related to a problem?

When printing walls with more than one line the first line is printed, then the nozzle makes a 90° turn, travels a short distance, makes another 90° turn and then prints the next line. These two 90° turns lead to short interruptions of the movement of the nozzle which in turn leads to a little blob of material. This doesn't look pretty and the additional material sometimes builds up over several layers.
The travel move between the 90° turns also leads to visible dents or protrusions in the wall surface.
Countless forum threads exist around this kind of wall blemishes.

### Describe the solution you'd like

I would like to at least have the option to replace the two 90° turns with an S-curve or a diagonal move that stretches the changeover from wall line to wall line out over a configurable distance. I imagine that this would look less obtrusive than either a scar or thousands of little bumps all over the surface.

Maybe Cura 5.x's ability to dynamically change wall line width could make it even possible to fade over from one wall line to the next one without any over or under extrusion.

### Describe alternatives you've considered

All other strategies to mitigate the visible consequences of the nozzle moving between wall lines only solve some of the issue in some circumstances.

### Affected users and/or printers

Cura users

MariMakes, 2022-08-29:
Hey @Triangulix,

I'm really trying to understand what problem you are encountering but my head keeps thinking this when you mention two 90° turns.

[image]

We have Combing and Jerk settings that can tune travel moves. 

Can you please add a drawing to describe your problem? 
That makes it easier to bring it up to the team. 
Thank you! 😄

Triangulix, 2022-08-31:
Hi @MariMakes,

you are right, it is not two times 90°, it is two times 45°.

This movement creates part of the artefact that we can either spread over the whole surface as blobs or dents or compact into one z seam. 
My proposal is to use the new variable line width capabilities of Cura to fade over from one wall line to the other.

[image]
[image]