Ultimaker/Cura #13023: Cant Download Plugin

Erox77, 2022-08-13:
### Application Version

5.1

### Platform

win 10

### Printer

biqu b1

### Reproduction steps

I tried to download polugin but it fails everytime
i've already re installed the program, tried to download a previous versions and i wasnt able to fix this problem

### Actual results

still cant downlaod

### Expected results

i hope i can download any desired plugin from the market

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9331513/cura.log)

please help, im new in 3d printing and its so frustrating.

GregValiant, 2022-08-13:
Thanks for the report.
Many of the plugins are written and maintained by contributors to Cura.  Which plugin are you having trouble with?

Erox77, 2022-08-13:
I'm not having trouble with any plugin, my problem is downloading them... i'm not able to download any plugin from the market but the plugin im trying to download is "custom cylinder suport" i was looking the Log file and i found it 

2022-08-13 10:23:35,600 - ERROR - [MainThread] Marketplace.PackageList._downloadError [251]: Failed to download package: CustomSupportCylinder due to <?xml version='1.0' encoding='UTF-8'?><Error><Code>ExpiredToken</Code><Message>The provided token has expired.</Message><Details>Request signature expired at: 2022-08-13T13:22:10+00:00</Details></Error>

i dont have any clue how to solve it

Piezoid, 2022-08-13:
Seems to be an issue with the API token:
> `The server returned: b"<?xml version='1.0' encoding='UTF-8'?><Error><Code>ExpiredToken</Code><Message>The provided token has expired.</Message><Details>Request signature expired at: 2022-08-13T13:24:11+00:00</Details></Error>`

The token renewal thread appears to be stuck:
> 2022-08-13 10:28:42,007 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [64]: Refreshing the access token for [https://account.ultimaker.com]
2022-08-13 10:28:42,008 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.refreshAccessToken [174]: Was already busy refreshing token. Do not start a new request.

@Erox77 Are you logged in with an Ultimaker account? Maybe retry the download after logging out?

Erox77, 2022-08-13:
yes im logged in a ultimaker account... ill try it now...

Erox77, 2022-08-13:
@Piezoid didnt work out... it seems a problem with the program... dont know what im gonna do

MariMakes, 2022-08-16:
Hey Erox77, 

You don't have to be signed in to download the plug-in. 
I attached a video showing how you can download the plug-in without being signed in.

https://user-images.githubusercontent.com/40423138/184848319-e928d42d-d548-4fee-9c2d-7f11eab76c0a.mp4

If that fails, you can visit this link and download the .curapackage file. (You will need the one with V8.0.0)
If you drag the download into Cura, and restart Cura your plug-in should be installed. 
https://github.com/5axes/CustomSupportCylinder/releases/tag/V2.6.4

Good luck! 
Can you let us know what if you are successful?

Erox77, 2022-08-16:
@MariMakes Thankkkk you very very much, the second option worked, i've already tried the first option trying to download unlogged, but it didnt work out, but i still can't download direct from the market . how can i find the plugins i want to do the same process ? i mean the plugin file? thank you again. 
[image]

MariMakes, 2022-08-16:
Hey @Erox77,

All plug-ins approved for Ultimaker Cura can be found on the Marketplace.
https://marketplace.ultimaker.com/app/cura/plugins

They all mention their github website where you can find their latest releases. 
You can try it with the Settings Guide, it's my favorite!

https://user-images.githubusercontent.com/40423138/184874301-89c198c5-81b8-45d9-b0de-2fa655a3420b.mp4

closed 2022-09-10 by github-actions[bot]

DarkDaskin, 2023-02-07:
I have the same problem for several months. Probably it started as soon as I upgraded to Cura 5.x.

From the **cura.log**:
`2023-02-07 18:13:03,000 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [369]: request[eeda666e][get][PyQt6.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/cura/v8.0.0/packages/SettingsGuide2/download')][timeout=None][no-data] got an QNetworkReplyError Error transferring https://storage.googleapis.com/um-cloud-production-cura-packages/v1/packages/SettingsGuide2/VclJULVQPP2g7gBmO33Ljg1TJPdF0lLcx8G8XD3-lstb/SettingsGuide2-v8.0.0-2022-11-02T16%3A37%3A52Z.curapackage?X-Goog-Algorithm=GOOG4-RSA-SHA256&X-Goog-Credential=cura-packages-api%40um-cloud-production.iam.gserviceaccount.com%2F20230207%2Fauto%2Fstorage%2Fgoog4_request&X-Goog-Date=20230207T151138Z&X-Goog-Expires=30&X-Goog-SignedHeaders=host&X-Goog-Signature=0d4757b8c005478fd0b41a06f5b614725c16333dc9555ec381575023999bcdb09d101418131fe00c0238144f13f3092c5a74c352737080c62974a5bc05bcef2dc30caa6be966f13544819ffd2d59938359e32ae6c01e1e95e6b6570b01615156584921e75e793c8783aa4d02daea4ead2df6b0e41d103d1d0330a2fe13eb2d293efad90bd52e7a7f20f171f1924020c386015bd729d3baabe2cbb6cfb98c1b384fdb94ad7a21fada5346b78a2cd5b90bbfc05252b4f6d61691dd3a209f8d19e1e861c701b21a5413b1889596790252ebb18808dce0bee6c91faa1988d1ff645dd54e428419498b00b9d191eb26b4594011caaab91a79e59395b854d35c9e117b - server replied: . The server returned: b"<?xml version='1.0' encoding='UTF-8'?><Error><Code>ExpiredToken</Code><Message>The provided token has expired.</Message><Details>Request signature expired at: 2023-02-07T15:12:08+00:00</Details></Error>"`

Indeed, opening that signed **storage.googleapis.com** URL in browser gives the same error. However, opening the previous **api.ultimaker.com** URL successfully downloads the curapackage file. That URL redirects to `https://storage.googleapis.com/um-cloud-production-cura-packages/v1/packages/SettingsGuide2/VclJULVQPP2g7gBmO33Ljg1TJPdF0lLcx8G8XD3-lstb/SettingsGuide2-v8.0.0-2022-11-02T16%3A37%3A52Z.curapackage?X-Goog-Algorithm=GOOG4-RSA-SHA256&X-Goog-Credential=cura-packages-api%40um-cloud-production.iam.gserviceaccount.com%2F20230207%2Fauto%2Fstorage%2Fgoog4_request&X-Goog-Date=20230207T151741Z&X-Goog-Expires=30&X-Goog-SignedHeaders=host&X-Goog-Signature=43f7d2619ba530ae57d34c072f4715543fe69d5f53b2bf6dd7850bacc4c56671aa122169720865e1ce1c102f93936787f6426b313eab8d5362c56312c7fdfcb6b716843ab77deb8a4b4854cbf40849cff67a7ea8e5cc21a9858c3d3b6cf4c81eca02c5864af18a3a4e4bf9515b9d4ff4b7fc18c367d2103989d9d0b737c15fd00cc599bd2e110e09bcbeac774f80868d9a7b61bdde6fabd9ace3817c7cdd684640f2d3395ccad943bc6541f6c639936581eaaa911f469bfcd19eb3d8e13cf1cebb1224f4ec33ce8cb17ad7a1d88c7017287d0498352278f919382e75e9f6f7eddc08cbfbf2296d2cf8bef1bc0029f54f169fe47c9b5625dea7d97dface51a07d`. Looks almost the same, but the `X-Goog-Date` parameter is different and equals to current date and time, while the same parameter from the log file is behind by few minutes, that's why it is expired. 
It looks that Cura waits for 1 minute before sending that request, so it expires. 

From WireShark:
[image]

Looks like this has nothing to do with Ultimaker account.

DarkDaskin, 2023-02-07:
On the other hand, using VPN prevents the issue, so it might be a network problem.

GregValiant, 2023-02-07:
There have been issues with Cura and some VPN's.  If you search here you may come up with some information.