Ultimaker/Cura #13026: Input fields that refresh the menu loose focus

Triangulix, 2022-08-13:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Input fields that change the visibility of other fields and therefore refresh the menu loose focus when they are emptied or filled.

This leads to the situation that changing Infill Density from 10 to 20 involves clicking into the field twice or sometimes even three times. After deleting the 10 the field has lost focus. After another click I can enter the 2 which immediately makes the field loose focus again. Only after a third click into the field I can type the 0 and have my 20% infill.

### Actual results

Described above

### Expected results

Standard Windows controls are fine. You don't have to re-invent every input box and repeat every mistake Microaoft made in 30 years. Same thing as the current scroll wheel issues.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

RenegadeZed, 2022-08-13:
i was googling for something similar, im not sure if its related. when you double click in the fields to resize or reposition the model on the build plate, you lose focus of the field and anything you type isnt registered.. then you press backspace and delete your model from the build plate.

Vandresc, 2022-08-16:
Hi @Triangulix, we have a ticket on our backlog regarded to this issue. Small but very annoying. I will close this ticket as a duplicate of https://github.com/Ultimaker/Cura/issues/12949, please follow any progress in the linked ticket. Thank you for your feedback!

closed 2022-08-16 by Vandresc

rdfry, 2023-01-26:
Still an issue in 5.2.2