Ultimaker/Cura #13027: Print Sequence: One at a Time becomes unslicable due to Z Height being shrunk too much

RealDanCarroll, 2022-08-13:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Crealty CR-10V3

### Reproduction steps

Load a model 
Multiply model by 3 so there are 4 copies (Number of copies doent matter as long as build plate isnt too crowded)
Seperate parts, 
change Print Sequence to "One at a time"

### Actual results

The Z Height shrinks from 400mm to about 10mm so most models are too tall to begin slicing.

### Expected results

I should be able to slice 4 small parts in 4 corners and not have the Z Space reduce to below the top of the models.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CantSlice.zip](https://github.com/Ultimaker/Cura/files/9331824/CantSlice.zip)

GregValiant, 2022-08-13:
Thanks for the report.
The X beam (the "Gantry") of most printers is a crash point if models can't fit underneath it as the bed and/or printhead moves in the Y.  It can only occur during "One at a Time" printing.
The height of the workspace is governed by the Gantry Height.  You will find it in the Print Head settings on the Machine Settings page.
This is from the CR-10 definition file that was submitted by someone within the community for inclusion in Cura.
"gantry_height": { "value": 25 }
It is the same as on my Ender 3 Pro and is the distance from the tip of the nozzle to the bottom of the X beam.
I'll close this as it is expected behavior and is there as a safety.

RealDanCarroll, 2022-08-13:
That doesnt sound like expected behavior to me. There is no reason it cant print all 4 of these models. If you need to move in the Y for some reason, raise to the highest Z previously achieved.

Ive printed these EXACT models One at a time on this exact machine with a previous version of cura.

GregValiant, 2022-08-14:
There is no bug here.  I've explained where the setting is located and I have explained why it is set to 25.  That would seem to be sufficient for you to figure out what you need to do.

RealDanCarroll, 2022-08-14:
You haven’t explained anything in any sufficient manner. You either don’t understand what my problem is, or you don’t understand how to communicate clearly. 

Regardless, you shouldn’t be tasked with responding to bug reports because you represent the project poorly.

Either explain how your the Cura UI and functionality has failed to communicate clearly or refrain from closing bug reports. I should be able to print 4 models in 4 separate corners of my bed and select “One at a time” without the app restricting me to such small Z height.

GregValiant, 2022-08-14:
As I said - in "Manage Printers" / "Machine Settings" look on the right at the "Print Head" settings.  One is "Gantry Height".  Change it to 250 or 400 or whatever.  It just needs to be higher than your models.  
[image]

Vandresc, 2022-08-16:
Hi @RealDanCarroll, @GregValiant reply provides the necessary/correct information, for safety reasons the gantry height is indeed reduced. You are also right that in previous versions (for example 4.13) you could have printed without being blocked but that is because the feature wasn't working completely. We did fix it and works as intended in the current version. This could be turned into a feature request if you wish but honestly, I am not very sure it will make it high on our priority list as the workaround is easy to perform compared with the rest of the critical issues we are trying to address. I hope you understand, 
thank you!

MariMakes, 2023-05-24:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue, and to me, it doesn't seem like a bug in UltiMaker Cura. 

Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-05-24 by MariMakes