Ultimaker/Cura #13029: Cura running extremely slow!

rodrigopbarreto, 2022-08-13:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Creality CR-10 V3

### Reproduction steps

Everytime I try to rescale or even to slice an STL the program gets unresponsive or extremely slow. It takes a lot to respond and eventually crashes. I've already tried to disable the UM3 Network Printer package with no succes. It's frustrating and it driving me to consider to stop using Cura as my primary slicer.

### Actual results

The program stops responding for several seconds and eventually crashes.

### Expected results

Everything should run smoothly. My PC has a decent configuration and when I triy the same slicing procces in another software it works fine.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[remote-support_22-08-13-16_27_20.zip](https://github.com/Ultimaker/Cura/files/9332059/remote-support_22-08-13-16_27_20.zip)

closed 2022-09-10 by github-actions[bot]

j-snijder, 2022-11-11:
Same here, since v 5.1.0

Cura is slower then a snail on frozen ice, not funny anymore.

That while Cura is/was the best out there :(

PeEd3d, 2022-12-01:
Hi,

I solved this Problem (for me) by switching the 3d acceleration settings to the Nvidia/Ati Card.

Standard Profile was the Processor Cpu inbuild Card 

I Hope this could help

j-snijder, 2022-12-01:
> Hi,
> 
> I solved this Problem (for me) by switching the 3d acceleration settings to the Nvidia/Ati Card.
> 
> Standard Profile was the Processor Cpu inbuild Card
> 
> I Hope this could help

That would mean that your other programm should act slower also, in my case thats not the issue.
As I work with both Creality slicer (if cura might fail, as it is right now) which has a cura version 4.8.1 as its base, and that one works fine.
But thanks for trying to help me.

In one of the other posts I am currently reporting and testing what happens, looks like ther is a problem with cura not cleaning up has data after a slice is completed ?
Still testing.