Ultimaker/Cura #13030: Add option for number of skirt layers

cmgreyhounds, 2022-08-13:
### Is your feature request related to a problem?

When I use a skirt in Cura the only way I can typically remove it from the build plate is by getting the edge of a small knife or razor blade underneath it to lift it up can pull it loose.

### Describe the solution you'd like

If we could use 1-3 layers for the skirt, then it would be tall enough to grab.
This is a feature of PrusaSlicer and it's extremely nice to have.

### Describe alternatives you've considered

None

### Affected users and/or printers

All

Asterchades, 2022-08-14:
The Draft Shield (Experimental) may do what you're after. If you configure the X/Y Distance to match your skirt offset, and set it to be Limited to the height you're after (eg 0.6mm for three layers at 0.2mm) it should build on top of the skirt. It will only be one wall wide, even if your skirt is wider, but in theory it should function similarly.

MariMakes, 2022-09-21:
Hey @cmgreyhounds,

Thanks for the request! 😄 
I personally really like @Asterchades suggestion. 
That is indeed the workaround that I use when have a need for something like this.

I brought your suggestion up to our process experts and they plan to do some research into this. 
For internal reference PP-231